open-power/hostboot #145: linker failing on ppc64le

shenki, 2018-08-20:
With op-build master (c57597296f58a6a2354e909fc81d6e3b4686bb8d) the linker step never exits. If I attach to one of the linker processes:

```
sudo strace -p `pgrep linker`
write(1, "Success\n", 8)                = 8
write(3, "securerom.elf,0x0\n", 18)     = 18
write(1, "Error writing to output.\n", 25) = 25
write(1, "Success\n", 8)                = 8
write(3, "securerom.elf,0x0\n", 18)     = 18
write(1, "Error writing to output.\n", 25) = 25
write(1, "Success\n", 8)                = 8
^Cwrite(3, "securerom.elf,0x0\n", 18)     = 18
```
Printing this sequence continually.

Taking a look at the spot in the build process between when it's succeeding and failing:

```
write(1, "\tSymbol: __cxa_guard_acquire\n", 29	Symbol: __cxa_guard_acquire
) = 29
write(1, "\t\tAddress: 178\n", 15		Address: 178
)      = 15
write(1, "\t\tGLOBAL\n", 9		GLOBAL
)             = 9
write(1, "\t\tFUNCTION\n", 11		FUNCTION
)          = 11
write(1, "\n", 1
)                       = 1
_llseek(4, 0, [0], SEEK_CUR)            = 0
_llseek(4, 0, [0], SEEK_CUR)            = 0
fstat(4, {st_mode=S_IFREG|0664, st_size=0, ...}) = 0
_llseek(4, 12288, [12288], SEEK_CUR)    = 0
write(4, "|B\23x|@\0\246xB\10@xB\370\2|[\3\2468@0\34|Z\3\246| \4\254"..., 4096) = 4096
write(4, "UJ\6\276\231I\0@| \4\254\201$\0\0a)\0\20\221$\0\0\201#\0\30/\211\0\0"..., 3332) = 3332
_llseek(4, 16384, [16384], SEEK_SET)    = 0
read(4, "UJ\6\276\231I\0@| \4\254\201$\0\0a)\0\20\221$\0\0\201#\0\30/\211\0\0"..., 4096) = 3332
_llseek(4, 4, [19720], SEEK_CUR)        = 0
write(4, "\27\10 \21INVALID_SECTION\0\0\0\0\0\0\0\0\0\0\0M0"..., 1336) = 1336
_llseek(4, 20480, [20480], SEEK_SET)    = 0
read(4, "\0\0\0\0\0\0>\260\0\0\0\0\0\0\315\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0>\314"..., 4096) = 576
write(3, "hbibl.elf,0x0\n", 14)         = 14
write(1, "Error writing to output.\n", 25Error writing to output.
) = 25
write(1, "Success\n", 8Success
)                = 8
write(3, "hbibl.elf,0x0\n", 14)         = 14
write(1, "Error writing to output.\n", 25Error writing to output.
) = 25
write(1, "Success\n", 8Success
)                = 8
write(3, "hbibl.elf,0x0\n", 14)         = 14
write(1, "Error writing to output.\n", 25Error writing to output.
) = 25
write(1, "Success\n", 8Success
)                = 8
write(3, "hbibl.elf,0x0\n", 14)         = 14
write(1, "Error writing to output.\n", 25Error writing to output.
) = 25
write(1, "Success\n", 8)                = -1 EPIPE (Broken pipe)
```

This is using Ubuntu 18.10 on ppc64le.

sathnaga, 2018-08-30:
op-build master(cc3dcdc55fd36c1bb2d13519a5b187a2d443a843) fails during hostboot compilation as it gets stuck during the linker stage and fills up the entire disk

Steps:
```
#git clone https://github.com/open-power/op-build
#cd op-build
# git submodule init && git submodule update
# . op-build-env
# op-build habanero_defconfig && op-build 
...
...
    LD         hbicore.elf^M
    LD         hbicore_test.elf^M
    LD         hbirt.elf^M
    LD         hbirt_test.elf^M
    LINKER     hbirt_test.bin^M
    LINKER     hbirt.bin^M
    LINKER     hbicore_test.bin^M
    LINKER     hbicore.bin^M
^M
bzip2: I/O or other error, bailing out.  Possible reason follows.^M
bzip2: No space left on device^M
        Input file = (stdin), output file = (stdout)^M
make[4]: *** [../src/build/mkrules/images.rules.mk:53: ../img/hbicore_test.bin] Error 1^M
make[4]: *** Waiting for unfinished jobs....^M
^M
bzip2: I/O or other error, bailing out.  Possible reason follows.^M
bzip2: No space left on device^M
        Input file = (stdin), output file = (stdout)^M
make[4]: *** [../src/build/mkrules/images.rules.mk:53: ../img/hbirt_test.bin] Error 1^M
^M
bzip2: I/O or other error, bailing out.  Possible reason follows.^M
bzip2: No space left on device^M
        Input file = (stdin), output file = (stdout)^M
make[4]: *** [../src/build/mkrules/images.rules.mk:53: ../img/hbirt.bin] Error 1^M
u -sh *^M
```

```
du -sh output/build/hostboot-p8-d3025f5d7ddd0723946bb54fcb471d2bf1fd2da4/
57G     output/build/hostboot-p8-d3025f5d7ddd0723946bb54fcb471d2bf1fd2da4/
```

Host Env:
```
P8 Tuleta with Fedora 28 base
# uname -r
4.18.0-g6e61beb7

# gcc -v
Using built-in specs.
COLLECT_GCC=gcc
COLLECT_LTO_WRAPPER=/usr/libexec/gcc/ppc64le-redhat-linux/8/lto-wrapper
OFFLOAD_TARGET_NAMES=nvptx-none
OFFLOAD_TARGET_DEFAULT=1
Target: ppc64le-redhat-linux
Configured with: ../configure --enable-bootstrap --enable-languages=c,c++,fortran,objc,obj-c++,ada,go,lto --prefix=/usr --mandir=/usr/share/man --infodir=/usr/share/info --with-bugurl=http://bugzilla.redhat.com/bugzilla --enable-shared --enable-threads=posix --enable-checking=release --enable-targets=powerpcle-linux --disable-multilib --with-system-zlib --enable-__cxa_atexit --disable-libunwind-exceptions --enable-gnu-unique-object --enable-linker-build-id --with-gcc-major-version-only --with-linker-hash-style=gnu --enable-plugin --enable-initfini-array --with-isl --disable-libmpx --enable-offload-targets=nvptx-none --without-cuda-driver --enable-gnu-indirect-function --enable-secureplt --with-long-double-128 --with-cpu-32=power8 --with-tune-32=power8 --with-cpu-64=power8 --with-tune-64=power8 --build=ppc64le-redhat-linux
Thread model: posix
gcc version 8.1.1 20180712 (Red Hat 8.1.1-5) (GCC) 

```

madscientist159, 2019-04-18:
We might be seeing the same underlying problem trying to update the Talos II / Blackbird firmware, but it's showing somewhat different symptoms on latest GIT master:

```
listdeps.pl ./img  -v
    MAKE       src IMAGE
    CSCOPE
    CTAGS
    LINKER     hbibl.bin
    LINKER     hbicore.bin
    LINKER     hbicore_test.bin
    LINKER     hbirt.bin
    LINKER     hbirt_test.bin
    LINKER     securerom.bin
26+0 records in
26+0 records out
26 bytes copied, 0.000158844 s, 164 kB/s
exception caught: Expected 1366965 found 2243509 at 2374040
../src/build/mkrules/images.rules.mk:53: recipe for target '../img/hbirt.bin' failed
make[4]: *** [../img/hbirt.bin] Error 255
make[4]: *** Waiting for unfinished jobs....
exception caught: Expected 1366965 found 2243509 at 2374040
../src/build/mkrules/images.rules.mk:53: recipe for target '../img/hbirt_test.bin' failed
make[4]: *** [../img/hbirt_test.bin] Error 255
exception caught: Expected 371 found 1074015803 at 290588
../src/build/mkrules/images.rules.mk:53: recipe for target '../img/hbicore.bin' failed
make[4]: *** [../img/hbicore.bin] Error 255
exception caught: Expected 371 found 1074015803 at 290588
../src/build/mkrules/images.rules.mk:53: recipe for target '../img/hbicore_test.bin' failed
make[4]: *** [../img/hbicore_test.bin] Error 255
../src/build/mkrules/passes.rules.mk:105: recipe for target '_BUILD/PASSES/IMAGE/BODY' failed
make[3]: *** [_BUILD/PASSES/IMAGE/BODY] Error 2
src/build/mkrules/passes.rules.mk:105: recipe for target '_BUILD/SUBDIR/IMAGE/src' failed
make[2]: *** [_BUILD/SUBDIR/IMAGE/src] Error 2
make[2]: *** Waiting for unfinished jobs....
src/build/mkrules/passes.rules.mk:105: recipe for target '_BUILD/PASSES/IMAGE/POST' failed
make[1]: *** [_BUILD/PASSES/IMAGE/POST] Error 2
make[1]: Leaving directory '/TALOS_FW_DEV/pnor/talos-op-build/output/build/hostboot-e8ef2bbe5c9bc999249163146c74d1889bbf39fd'
package/pkg-generic.mk:238: recipe for target '/TALOS_FW_DEV/pnor/talos-op-build/output/build/hostboot-e8ef2bbe5c9bc999249163146c74d1889bbf39fd/.stamp_built' failed
make: *** [/TALOS_FW_DEV/pnor/talos-op-build/output/build/hostboot-e8ef2bbe5c9bc999249163146c74d1889bbf39fd/.stamp_built] Error 2
make: Leaving directory '/TALOS_FW_DEV/pnor/talos-op-build/buildroot'
```

madscientist159, 2019-04-18:
It looks like this was fixed in commit 9be05f40, should this report be closed?

dcrowell77, 2019-04-18:
Sounds right to me, reopen if something is still broken.

closed 2019-04-18 by dcrowell77