open-power/hostboot #153: Help decoding RAM replacement error:   Failed due to wrong DIMM type

seanbruno, 2018-12-03:
I'm a bit unclear what the real problem here is.  I'm suspicious that this Tyan SP010 has a hardcoded list of RAM modules that are allowed and that I have chosen a set that isn't in the "whitelist" or whatever.  Can someone help me decode what the actual error is here?  

  3.09862|Ignoring boot flags, incorrect version 0x0
  3.16640|ISTEP  6. 3 - host_init_fsi
  3.53840|ISTEP  6. 4 - host_set_ipl_parms
  3.53940|ISTEP  6. 5 - host_discover_targets
  7.69413|HWAS|PRESENT> DIMM[03]=00AA000000000000
  7.69413|HWAS|PRESENT> Membuf[04]=4000000000000000
  7.69414|HWAS|PRESENT> Proc[05]=8000000000000000
  7.81097|ISTEP  6. 6 - host_gard
  7.94620|ISTEP  6. 7 - host_cancontinue_clear
  9.81919|ISTEP  6. 8 - proc_revert_sbe_mcs_setup
  9.86738|ISTEP  6. 9 - host_slave_sbe_config
 10.84801|ISTEP  6.10 - host_sbe_start
 10.84859|ISTEP  6.11 - proc_check_slave_sbe_seeprom_complete
 10.96844|ISTEP  6.12 - proc_cen_ref_clk_enable
 10.98830|ISTEP  6.13 - proc_xmit_sbe
 10.98901|ISTEP  7. 1 - proc_attr_update
 11.05714|ISTEP  7. 2 - proc_a_x_pci_dmi_pll_initf
 11.18373|ISTEP  7. 3 - proc_a_x_pci_dmi_pll_setup
 11.21369|ISTEP  7. 4 - proc_startclock_chiplets
 11.25285|ISTEP  7. 5 - proc_chiplet_scominit
 11.45001|ISTEP  7. 6 - proc_xbus_scominit
 11.48871|ISTEP  7. 7 - proc_abus_scominit
 11.49892|ISTEP  7. 8 - proc_pcie_scominit
 11.58869|ISTEP  7. 9 - proc_scomoverride_chiplets
 11.58940|ISTEP  8. 1 - fabric_erepair
 11.64792|ISTEP  8. 2 - fabric_io_dccal
 11.64872|ISTEP  8. 3 - fabric_pre_trainadv
 11.66080|ISTEP  8. 4 - fabric_io_run_training
 11.66154|ISTEP  8. 5 - fabric_post_trainadv
 11.66498|ISTEP  8. 6 - host_startprd_pbus
 12.26419|ISTEP  8. 7 - host_attnlisten_proc
 12.26495|ISTEP  8. 8 - proc_fab_iovalid
 12.30430|ISTEP  9. 1 - proc_build_smp
 12.65467|ISTEP  9. 2 - host_slave_sbe_update
 13.19619|ISTEP 10. 1 - host_prd_hwreconfig
 13.45697|ISTEP 10. 2 - cen_sbe_tp_chiplet_init1
 13.96605|ISTEP 10. 3 - cen_sbe_pll_initf
 13.96672|ISTEP 10. 4 - cen_sbe_pll_setup
 13.96741|ISTEP 10. 5 - cen_sbe_tp_chiplet_init2
 13.96826|ISTEP 10. 6 - cen_sbe_tp_arrayinit
 13.96905|ISTEP 10. 7 - cen_sbe_tp_chiplet_init3
 13.97095|ISTEP 10. 8 - cen_sbe_chiplet_init
 13.97164|ISTEP 10. 9 - cen_sbe_arrayinit
 13.97227|ISTEP 10.10 - cen_sbe_dts_init
 13.97456|ISTEP 10.11 - cen_sbe_initf
 13.97523|ISTEP 10.12 - cen_sbe_do_manual_inits
 13.97596|ISTEP 10.13 - cen_sbe_nest_startclocks
 13.97683|ISTEP 10.14 - cen_sbe_scominits
 13.97745|ISTEP 11. 1 - mss_getecid
 14.16865|ISTEP 11. 2 - dmi_attr_update
 14.16932|ISTEP 11. 3 - proc_dmi_scominit
 14.35470|ISTEP 11. 4 - dmi_scominit
 14.39396|ISTEP 11. 5 - dmi_erepair
 14.52361|ISTEP 11. 6 - dmi_io_dccal
 14.91786|ISTEP 11. 7 - dmi_pre_trainadv
 14.91868|ISTEP 11. 8 - dmi_io_run_training
 15.05353|ISTEP 11. 9 - dmi_post_trainadv
 15.05455|ISTEP 11.10 - proc_cen_framelock
 15.11698|ISTEP 11.11 - host_startprd_dmi
 15.11770|ISTEP 11.12 - host_attnlisten_cen
 15.11851|ISTEP 11.13 - cen_set_inband_addr
 15.11932|ISTEP 12. 1 - host_collect_dimm_spd
 15.22509|ISTEP 12. 2 - mss_volt
 15.32379|ISTEP 12. 3 - mss_freq
 15.35351|ISTEP 12. 4 - mss_eff_config
 15.44944|================================================
 15.45902|Error reported by hwpf (0x0900)
 15.46906|  Failed due to wrong DIMM type
 15.46907|  ModuleId   0x22 MOD_GET_RCD_CNTL_WORD
 15.46907|  ReasonCode 0x0933 RC_INVALID_DIMM_TYPE
 15.46908|  UserData1  DIMM TYPE : 0x000000000000000b
 15.47866|  UserData2   : 0x0000000000000000
 15.47866|User Data Section 0, type UD
 15.47867|  Subsection type 0x06
 15.47867|  ComponentId errl (0x0100)
 15.47867|  CALLOUT
 15.47867|  PROCEDURE ERROR
 15.47868|  Procedure: 85
 15.47868|User Data Section 1, type UD
 15.47868|  Subsection type 0x15
 15.47869|  ComponentId hb-trace (0x3100)
 15.47869|User Data Section 2, type UD
 15.47869|  Subsection type 0x15
 15.47870|  ComponentId hb-trace (0x3100)
 15.47870|User Data Section 3, type UD
 15.47870|  Subsection type 0x15
 15.47870|  ComponentId hb-trace (0x3100)
 15.47871|User Data Section 4, type UD
 15.47871|  Subsection type 0x02
 15.47871|  ComponentId errl (0x0100)
 15.47872|  TARGET
 15.47872|  HUID: 000d0000
 15.47874|  Phys path: Physical:/Sys0/Node0/Membuf1/MBA0
 15.47874|  Affinity path: Logical:/Sys0/Node0/Proc0/MCS1/Membuf1/MBA0
 15.47875|User Data Section 5, type UD
 15.47875|  Subsection type 0x15
 15.47875|  ComponentId hb-trace (0x3100)
 15.47876|User Data Section 6, type UD
 15.47876|  Subsection type 0x03
 15.47876|  ComponentId errl (0x0100)
 15.47877|User Data Section 7, type UD
 15.47877|  Subsection type 0x01
 15.47877|  ComponentId errl (0x0100)
 15.47877|  STRING
 15.47878|  Hostboot Build ID: 
 15.47878|================================================
 17.26950|================================================
 17.26951|Error reported by hwpf (0x0900)
 17.26951|  Failed due to wrong DIMM type
 17.26951|  ModuleId   0x22 MOD_GET_RCD_CNTL_WORD
 17.26952|  ReasonCode 0x0933 RC_INVALID_DIMM_TYPE
 17.26952|  UserData1  DIMM TYPE : 0x000000000000000b
 17.26953|  UserData2   : 0x0000000000000000
 17.26953|User Data Section 0, type UD
 17.26953|  Subsection type 0x06
 17.26954|  ComponentId errl (0x0100)
 17.26954|  CALLOUT
 17.26954|  PROCEDURE ERROR
 17.26954|  Procedure: 85
 17.26955|User Data Section 1, type UD
 17.26955|  Subsection type 0x15
 17.26956|  ComponentId hb-trace (0x3100)
 17.26956|User Data Section 2, type UD
 17.26956|  Subsection type 0x15
 17.26957|  ComponentId hb-trace (0x3100)
 17.26957|User Data Section 3, type UD
 17.26958|  Subsection type 0x15
 17.26958|  ComponentId hb-trace (0x3100)
 17.26959|User Data Section 4, type UD
 17.26959|  Subsection type 0x02
 17.26959|  ComponentId errl (0x0100)
 17.26960|  TARGET
 17.26960|  HUID: 000d0001
 17.26962|  Phys path: Physical:/Sys0/Node0/Membuf1/MBA1
 17.26963|  Affinity path: Logical:/Sys0/Node0/Proc0/MCS1/Membuf1/MBA1
 17.26963|User Data Section 5, type UD
 17.26964|  Subsection type 0x15
 17.26964|  ComponentId hb-trace (0x3100)
 17.26965|User Data Section 6, type UD
 17.26965|  Subsection type 0x03
 17.26965|  ComponentId errl (0x0100)
 17.26966|User Data Section 7, type UD
 17.26966|  Subsection type 0x01
 17.26967|  ComponentId errl (0x0100)
 17.26967|  STRING
 17.26967|  Hostboot Build ID: 
 17.26968|================================================
 17.68160|System shutting down with error status 0x900000AE
 17.70089|================================================
 17.70090|Error reported by istep (0x1700)
 17.70090|  IStep failed, see other log(s) with the same PLID for reason.
 17.70090|  ModuleId   0x01 ISTEP_REPORTING_ERROR
 17.70091|  ReasonCode 0x1701 ISTEP_FAILURE
 17.71053|  UserData1  eid of first error : 0x900000ae00000933
 17.71054|  UserData2  Reason code of first error : 0x0000000200000c04
 17.71054|User Data Section 0, type UD
 17.71054|  Subsection type 0x03
 17.71055|  ComponentId hwpf (0x0900)
 17.71055|User Data Section 1, type UD
 17.71055|  Subsection type 0x03
 17.71056|  ComponentId hwpf (0x0900)
 17.71056|User Data Section 2, type UD
 17.71056|  Subsection type 0x01
 17.71056|  ComponentId errl (0x0100)
 17.71057|  STRING
 17.71057|  mss_eff_config
 17.71057|User Data Section 3, type UD
 17.71058|  Subsection type 0x06
 17.71058|  ComponentId errl (0x0100)
 17.71058|  CALLOUT
 17.71058|  PROCEDURE ERROR
 17.71059|  Procedure: 85
 17.71059|User Data Section 4, type UD
 17.71059|  Subsection type 0x03
 17.71060|  ComponentId errl (0x0100)
 17.71060|User Data Section 5, type UD
 17.71060|  Subsection type 0x01
 17.71060|  ComponentId errl (0x0100)
 17.71061|  STRING
 17.71061|  Hostboot Build ID: 
 17.71061|================================================

stewart-ibm, 2018-12-03:
Reopening as I'm trying out the "move this issue to another repository" feature. Hostboot is possibly the right place to have this here?

seanbruno, 2018-12-03:
ooo, neat.

dcrowell77, 2018-12-03:
From the code (src/usr/hwpf/plat/fapiPlatAttributeService.C) it looks like it is checking for RDIMM or UDIMM support.  It looks like this dimm is a LRDIMM, which is not supported by this codebase.

closed 2018-12-03 by dcrowell77