open-power/hostboot #159: Tyan P8 hostboot DIMM detect failure.

seanbruno, 2018-12-14:
So, I *thought* I ordered compatible RAM for my SP-010 and I'm still getting a hwpf failure with the 32GB version of these modules:  https://www.skhynix.com/product/filedata/fileDownload.do?seq=3689

They appear to be backwards compatible, but they are definitely being rejected by hwpf here:

3.81237|Ignoring boot flags, incorrect version 0x0
  3.86996|ISTEP  6. 3 - host_init_fsi
  4.24179|ISTEP  6. 4 - host_set_ipl_parms
  4.24241|ISTEP  6. 5 - host_discover_targets
  6.17712|HWAS|PRESENT> DIMM[03]=00AA000000000000
  6.17713|HWAS|PRESENT> Membuf[04]=4000000000000000
  6.17713|HWAS|PRESENT> Proc[05]=8000000000000000
  6.29378|ISTEP  6. 6 - host_gard
  6.42918|ISTEP  6. 7 - host_cancontinue_clear
  8.30650|ISTEP  6. 8 - proc_revert_sbe_mcs_setup
  8.35468|ISTEP  6. 9 - host_slave_sbe_config
  9.33548|ISTEP  6.10 - host_sbe_start
  9.33614|ISTEP  6.11 - proc_check_slave_sbe_seeprom_complete
  9.45596|ISTEP  6.12 - proc_cen_ref_clk_enable
  9.47592|ISTEP  6.13 - proc_xmit_sbe
  9.47685|ISTEP  7. 1 - proc_attr_update
  9.54481|ISTEP  7. 2 - proc_a_x_pci_dmi_pll_initf
  9.67035|ISTEP  7. 3 - proc_a_x_pci_dmi_pll_setup
  9.70048|ISTEP  7. 4 - proc_startclock_chiplets
  9.73955|ISTEP  7. 5 - proc_chiplet_scominit
  9.93730|ISTEP  7. 6 - proc_xbus_scominit
  9.97599|ISTEP  7. 7 - proc_abus_scominit
  9.98618|ISTEP  7. 8 - proc_pcie_scominit
 10.06051|ISTEP  7. 9 - proc_scomoverride_chiplets
 10.06128|ISTEP  8. 1 - fabric_erepair
 10.13789|ISTEP  8. 2 - fabric_io_dccal
 10.13853|ISTEP  8. 3 - fabric_pre_trainadv
 10.14882|ISTEP  8. 4 - fabric_io_run_training
 10.14952|ISTEP  8. 5 - fabric_post_trainadv
 10.15377|ISTEP  8. 6 - host_startprd_pbus
 10.75476|ISTEP  8. 7 - host_attnlisten_proc
 10.75544|ISTEP  8. 8 - proc_fab_iovalid
 10.79437|ISTEP  9. 1 - proc_build_smp
 11.13951|ISTEP  9. 2 - host_slave_sbe_update
 11.64909|ISTEP 10. 1 - host_prd_hwreconfig
 12.08109|ISTEP 10. 2 - cen_sbe_tp_chiplet_init1
 12.59456|ISTEP 10. 3 - cen_sbe_pll_initf
 12.59533|ISTEP 10. 4 - cen_sbe_pll_setup
 12.59597|ISTEP 10. 5 - cen_sbe_tp_chiplet_init2
 12.59686|ISTEP 10. 6 - cen_sbe_tp_arrayinit
 12.59760|ISTEP 10. 7 - cen_sbe_tp_chiplet_init3
 12.59958|ISTEP 10. 8 - cen_sbe_chiplet_init
 12.60039|ISTEP 10. 9 - cen_sbe_arrayinit
 12.60290|ISTEP 10.10 - cen_sbe_dts_init
 12.60398|ISTEP 10.11 - cen_sbe_initf
 12.60507|ISTEP 10.12 - cen_sbe_do_manual_inits
 12.60609|ISTEP 10.13 - cen_sbe_nest_startclocks
 12.60714|ISTEP 10.14 - cen_sbe_scominits
 12.60818|ISTEP 11. 1 - mss_getecid
 12.78004|ISTEP 11. 2 - dmi_attr_update
 12.78065|ISTEP 11. 3 - proc_dmi_scominit
 12.96547|ISTEP 11. 4 - dmi_scominit
 13.00513|ISTEP 11. 5 - dmi_erepair
 13.13516|ISTEP 11. 6 - dmi_io_dccal
 13.51959|ISTEP 11. 7 - dmi_pre_trainadv
 13.52059|ISTEP 11. 8 - dmi_io_run_training
 13.65534|ISTEP 11. 9 - dmi_post_trainadv
 13.65611|ISTEP 11.10 - proc_cen_framelock
 13.71464|ISTEP 11.11 - host_startprd_dmi
 13.71530|ISTEP 11.12 - host_attnlisten_cen
 13.71614|ISTEP 11.13 - cen_set_inband_addr
 13.71685|ISTEP 12. 1 - host_collect_dimm_spd
 13.83231|ISTEP 12. 2 - mss_volt
 13.92784|ISTEP 12. 3 - mss_freq
 13.95733|ISTEP 12. 4 - mss_eff_config
 14.05594|================================================
 14.06598|Error reported by hwpf (0x0900)
 14.07555|  Failed due to wrong DIMM type
 14.07555|  ModuleId   0x22 MOD_GET_RCD_CNTL_WORD
 14.07556|  ReasonCode 0x0933 RC_INVALID_DIMM_TYPE
 14.07556|  UserData1  DIMM TYPE : 0x000000000000000b
 14.08512|  UserData2   : 0x0000000000000000
 14.08513|User Data Section 0, type UD
 14.08513|  Subsection type 0x06
 14.08513|  ComponentId errl (0x0100)
 14.08514|  CALLOUT
 14.08514|  PROCEDURE ERROR
 14.08514|  Procedure: 85
 14.08514|User Data Section 1, type UD
 14.08515|  Subsection type 0x15
 14.08515|  ComponentId hb-trace (0x3100)
 14.08515|User Data Section 2, type UD
 14.08516|  Subsection type 0x15
 14.08516|  ComponentId hb-trace (0x3100)
 14.08516|User Data Section 3, type UD
 14.08517|  Subsection type 0x15
 14.08517|  ComponentId hb-trace (0x3100)
 14.08517|User Data Section 4, type UD
 14.08518|  Subsection type 0x02
 14.08518|  ComponentId errl (0x0100)
 14.08518|  TARGET
 14.08518|  HUID: 000d0000
 14.08520|  Phys path: Physical:/Sys0/Node0/Membuf1/MBA0
 14.08521|  Affinity path: Logical:/Sys0/Node0/Proc0/MCS1/Membuf1/MBA0
 14.08521|User Data Section 5, type UD
 14.08521|  Subsection type 0x15
 14.08522|  ComponentId hb-trace (0x3100)
 14.08522|User Data Section 6, type UD
 14.08522|  Subsection type 0x03
 14.08523|  ComponentId errl (0x0100)
 14.08523|User Data Section 7, type UD
 14.08523|  Subsection type 0x01
 14.08524|  ComponentId errl (0x0100)
 14.08524|  STRING
 14.08524|  Hostboot Build ID: 
 14.08524|================================================
 15.95625|================================================
 15.95626|Error reported by hwpf (0x0900)
 15.95626|  Failed due to wrong DIMM type
 15.95626|  ModuleId   0x22 MOD_GET_RCD_CNTL_WORD
 15.95627|  ReasonCode 0x0933 RC_INVALID_DIMM_TYPE
 15.95627|  UserData1  DIMM TYPE : 0x000000000000000b
 15.95627|  UserData2   : 0x0000000000000000
 15.95628|User Data Section 0, type UD
 15.95628|  Subsection type 0x06
 15.95628|  ComponentId errl (0x0100)
 15.95628|  CALLOUT
 15.95629|  PROCEDURE ERROR
 15.95629|  Procedure: 85
 15.95629|User Data Section 1, type UD
 15.95629|  Subsection type 0x15
 15.95630|  ComponentId hb-trace (0x3100)
 15.95630|User Data Section 2, type UD
 15.95630|  Subsection type 0x15
 15.95631|  ComponentId hb-trace (0x3100)
 15.95631|User Data Section 3, type UD
 15.95631|  Subsection type 0x15
 15.95632|  ComponentId hb-trace (0x3100)
 15.95632|User Data Section 4, type UD
 15.95632|  Subsection type 0x02
 15.95633|  ComponentId errl (0x0100)
 15.95633|  TARGET
 15.95633|  HUID: 000d0001
 15.95635|  Phys path: Physical:/Sys0/Node0/Membuf1/MBA1
 15.95635|  Affinity path: Logical:/Sys0/Node0/Proc0/MCS1/Membuf1/MBA1
 15.95636|User Data Section 5, type UD
 15.95636|  Subsection type 0x15
 15.95636|  ComponentId hb-trace (0x3100)
 15.95637|User Data Section 6, type UD
 15.95637|  Subsection type 0x03
 15.95637|  ComponentId errl (0x0100)
 15.95638|User Data Section 7, type UD
 15.95638|  Subsection type 0x01
 15.95638|  ComponentId errl (0x0100)
 15.95638|  STRING
 15.95639|  Hostboot Build ID:
 15.95639|================================================
 16.46747|System shutting down with error status 0x9000011B
 16.48686|================================================
 16.48686|Error reported by istep (0x1700)
 16.48687|  IStep failed, see other log(s) with the same PLID for reason.
 16.48687|  ModuleId   0x01 ISTEP_REPORTING_ERROR
 16.48687|  ReasonCode 0x1701 ISTEP_FAILURE
 16.49660|  UserData1  eid of first error : 0x9000011b00000933
 16.49660|  UserData2  Reason code of first error : 0x0000000200000c04
 16.49661|User Data Section 0, type UD
 16.49661|  Subsection type 0x03
 16.49662|  ComponentId hwpf (0x0900)
 16.49662|User Data Section 1, type UD
 16.49662|  Subsection type 0x03
 16.49663|  ComponentId hwpf (0x0900)
 16.49663|User Data Section 2, type UD
 16.49663|  Subsection type 0x01
 16.49664|  ComponentId errl (0x0100)
 16.49664|  STRING
 16.49664|  mss_eff_config
 16.49665|User Data Section 3, type UD
 16.49665|  Subsection type 0x06
 16.49666|  ComponentId errl (0x0100)
 16.49666|  CALLOUT
 16.49666|  PROCEDURE ERROR
 16.49666|  Procedure: 85
 16.49667|User Data Section 4, type UD
 16.49667|  Subsection type 0x03
 16.49667|  ComponentId errl (0x0100)
 16.49668|User Data Section 5, type UD
 16.49668|  Subsection type 0x01
 16.49668|  ComponentId errl (0x0100)
 16.49669|  STRING
 16.49669|  Hostboot Build ID:
 16.49669|================================================

dcrowell77, 2019-01-02:
This is still showing that you are using a LRDIMM.  These are not supported.

seanbruno, 2019-01-02:
I think, my point and probably wrong, was that these DIMMs are *supposedly* backwards compatible.  At least that's what the manufacturer says.  I can accept, "no Sean, you're wrong" as an answer too.  :-)

dcrowell77, 2019-01-02:
Can you dump out the contents of the SPD?  The DJVPD pnor partition would have the data if that is simpler.  From looking at the code it seems like the only way we'd get here is if the SPD on the DIMM told us it was a LRDIMM (byte 3 = 0xB).

seanbruno, 2019-01-02:
I cannot dump this any longer as I returned the RAM to the seller.  So, probably we should just close this issue and I should try really hard to get supported RAM for this box.

ecorderoibm, 2019-01-02:
Hello Sean - can you let me know what was the exact part number they sent you and also what have you ordered to put in the machine next?

seanbruno, 2019-01-02:
HMT84GR7AMR4A-PB is the part I attempted to use.  (32GB).

ecorderoibm, 2019-01-02:
Great thank you. I found this : 
HYNIX HMT84GR7AMR4A-PB 32GB (1X32GB) 1600MHZ PC3-12800 CL11 ECC REGISTERED QUAD RANK 1.35V DDR3 SDRAM LOAD REDUCED 240-PIN DIMM MEMORY MODULE FOR SERVER MEMORY.
They were "Load Reduced", like Dan mentioned. We do not support those in P8. 
Let me know if you have any questions.

seanbruno, 2019-01-02:
The thing that threw me, is the "backwards compatible" nonsense.  Can you recommend a 32GB DIMM that should "just work" so I can stop harassing y'all?

ecorderoibm, 2019-01-02:
Sure. You're looking for 32GB DDR3?

seanbruno, 2019-01-02:
That would be the ideal.  We're trying to use x2 of these machines in the FreeBSD Project as pkg builders.  So, we do things like compile compilers all the time.  The 4 slots on the Tyan board, mean that my ram choices are limited.

ecorderoibm, 2019-01-02:
These are the DIMMs we qualified for a couple of machines most similar to your application. 

MICRON TECHNOLOGY INC	- MT72KSS4G72PZ-1G4N1
SAMSUNG SEMICONDUCTOR INC - M393B4G70DM0-YH9
SK HYNIX AMERICA INC	 -  HMT84GR7BMR4A-H9 T8/T4

seanbruno, 2019-01-02:
Ok, thank you for your guidance.  I'll give it a spin and see what happens.

closed 2019-01-02 by seanbruno

seanbruno, 2019-01-09:
> These are the DIMMs we qualified for a couple of machines most similar to your application.
> 
> MICRON TECHNOLOGY INC - MT72KSS4G72PZ-1G4N1
> SAMSUNG SEMICONDUCTOR INC - M393B4G70DM0-YH9
> SK HYNIX AMERICA INC - HMT84GR7BMR4A-H9 T8/T4

Thank you for the examples.  The Samsung DIMMs worked nicely and we are now able to run this server maxed out.