open-power/hostboot #172: Hostboot fails in linking stage on ppc64el systems

madscientist159, 2019-04-18:
When attempting to build hostboot on a Debian Stretch ppc64el system, the build fails with:

```
listdeps.pl ./img  -v
    MAKE       src IMAGE
    CSCOPE
    CTAGS
    LINKER     hbibl.bin
    LINKER     hbicore.bin
    LINKER     hbicore_test.bin
    LINKER     hbirt.bin
    LINKER     hbirt_test.bin
    LINKER     securerom.bin
26+0 records in
26+0 records out
26 bytes copied, 0.000158844 s, 164 kB/s
exception caught: Expected 1366965 found 2243509 at 2374040
../src/build/mkrules/images.rules.mk:53: recipe for target '../img/hbirt.bin' failed
make[4]: *** [../img/hbirt.bin] Error 255
make[4]: *** Waiting for unfinished jobs....
exception caught: Expected 1366965 found 2243509 at 2374040
../src/build/mkrules/images.rules.mk:53: recipe for target '../img/hbirt_test.bin' failed
make[4]: *** [../img/hbirt_test.bin] Error 255
exception caught: Expected 371 found 1074015803 at 290588
../src/build/mkrules/images.rules.mk:53: recipe for target '../img/hbicore.bin' failed
make[4]: *** [../img/hbicore.bin] Error 255
exception caught: Expected 371 found 1074015803 at 290588
../src/build/mkrules/images.rules.mk:53: recipe for target '../img/hbicore_test.bin' failed
make[4]: *** [../img/hbicore_test.bin] Error 255
../src/build/mkrules/passes.rules.mk:105: recipe for target '_BUILD/PASSES/IMAGE/BODY' failed
make[3]: *** [_BUILD/PASSES/IMAGE/BODY] Error 2
src/build/mkrules/passes.rules.mk:105: recipe for target '_BUILD/SUBDIR/IMAGE/src' failed
make[2]: *** [_BUILD/SUBDIR/IMAGE/src] Error 2
make[2]: *** Waiting for unfinished jobs....
src/build/mkrules/passes.rules.mk:105: recipe for target '_BUILD/PASSES/IMAGE/POST' failed
make[1]: *** [_BUILD/PASSES/IMAGE/POST] Error 2
make[1]: Leaving directory '/TALOS_FW_DEV/pnor/talos-op-build/output/build/hostboot-e8ef2bbe5c9bc999249163146c74d1889bbf39fd'
package/pkg-generic.mk:238: recipe for target '/TALOS_FW_DEV/pnor/talos-op-build/output/build/hostboot-e8ef2bbe5c9bc999249163146c74d1889bbf39fd/.stamp_built' failed
make: *** [/TALOS_FW_DEV/pnor/talos-op-build/output/build/hostboot-e8ef2bbe5c9bc999249163146c74d1889bbf39fd/.stamp_built] Error 2
make: Leaving directory '/TALOS_FW_DEV/pnor/talos-op-build/buildroot'
```

stewart-ibm, 2019-04-18:
Timothy Pearson <notifications@github.com> writes:
> When attempting to build hostboot on a Debian Stretch ppc64el system,
> the build fails with:

Are you using the toolchain native to debian stretch?

There's a "oh goodness, nobody wants to deal with that" known issue with
the hostboot linker where it doesn't deal with certain types of
relocations even though it should.

That being said, it looks like Luis has been doing some work fixing up
all of the compiler warnings (and failings like this) for Hostboot with
more recent GCC, so perhaps Luis can point you to the commits that fix
it, as certainly https://github.com/open-power/op-build/pull/2803 seems
to indicate that this may be something fixed in recent hostboot.

-- 
Stewart Smith
OPAL Architect, IBM.

madscientist159, 2019-04-18:
@stewart-ibm I'm compiling this from the latest hostboot master and a very recent op-build.  It's holding up firmware updates...do you have time for a quick chat on IRC?

madscientist159, 2019-04-18:
I can confirm https://github.com/open-power/op-build/pull/2803 resolves this issue

dcrowell77, 2019-04-18:
FYI @lpfluisfer

stewart-ibm, 2019-04-29:
Timothy Pearson <notifications@github.com> writes:
> I can confirm https://github.com/open-power/op-build/pull/2803
> resolves this issue

Ahh excellent.

Hopefully we can get that all upstream soon.

-- 
Stewart Smith
OPAL Architect, IBM.