open-power/hostboot #181: Bit of an explanation on this line.

hanetzer, 2019-08-19:
https://github.com/open-power/hostboot/blob/5825828fa7984e6d46dd47ef1835646303d2a593/src/kernel/start.S#L36
So, this line says it clears `MSR[TA] (bit 1)` but Power ISA Version 3.0 B pg. 943 states that
bits 1-2 of the MSR are 'reserved', and there is no bit I can see called `TA`. I was hoping
someone could elaborate on this, or let me know I'm interpreting it wrong.

npiggin, 2019-08-19:
You are reading the ISA correctly, these MSR bits are reserved.
It predates my involvement with OpenPOWER but the code or comment may have inadvertently come from some legacy code base or documentation.

dcrowell77, 2019-08-19:
Remember that the Power ISA AS permits implementation-specific extensions to the architecture.  At this point I can only tell you that bit 1 does need to be zero for Hostboot and OPAL.

dcrowell77, 2019-08-20:
FYI - https://www.techrepublic.com/article/open-source-power-isa-takes-aim-at-intel-and-arm-for-accelerator-driven-computing/

Things might be getting a bit more open going forward.

friedkiwi, 2019-08-31:
@hanetzer see https://archive.midrange.com/mi400/200502/msg00004.html for a good starting point. You might want to plow through that mailing list archive if you want to know more about tags active mode.

hlandau, 2019-08-31:
@hanetzer See here: [https://www.devever.net/~hl/ppcas](https://www.devever.net/~hl/ppcas)

hanetzer, 2019-08-31:
> @hanetzer see https://archive.midrange.com/mi400/200502/msg00004.html for a good starting
> point. You might want to plow through that mailing list archive if you want to know more about
> tags active mode.

Will do, thanks.

> @hanetzer See here: https://www.devever.net/~hl/ppcas

Thankya, I actually have this open in a tab at my home workstation.

friedkiwi, 2019-08-31:
@hanetzer: if you want to play with IBM i you can always pop into ##ibmi on freenode or on r/IBMiHUG. There’s a few semi-public machines people operate.