open-power/hostboot #188: PnorUtils.pm creates unnecessary file

artemsen, 2019-10-11:
Package PnorUtils.pm provides function `loadPnorLayout` that called from different places.
The commit 845fb4492dce998d95a0ab2542b05c866c0f5edc modifies that function and now every call to `loadPnorLoyout` creates a temporary file `*WithOffsets.xml` in current directory:
https://github.com/open-power/hostboot/blob/845fb4492dce998d95a0ab2542b05c866c0f5edc/src/build/buildpnor/PnorUtils.pm#L258

It's not a big problem, but after building we always have DIRTY flag reported by git, because script `buildpnor.pl` works in the default directory (`op-build/buildroot`), which now contains excess file.

lpfluisfer, 2019-10-11:
I'm taking a look at this. Ultimately we wanted those *WithOffsets.xml files to end up in: 
_<op-build_repo>/output/host/powerpc64le-buildroot-linux-gnu/sysroot/openpower_pnor_scratch/defaultPnorLayout_64WithOffsets.xml_

But it may be that the script is being run from the Buildroot level as well.

artemsen, 2019-10-11:
It actually ends up there, when PnorUtils.pm is called by [genPnorImages.pl](https://github.com/open-power/hostboot/blob/845fb4492dce998d95a0ab2542b05c866c0f5edc/src/build/buildpnor/genPnorImages.pl#L389).
But the second call from [buildpnor.pl](https://github.com/open-power/hostboot/blob/845fb4492dce998d95a0ab2542b05c866c0f5edc/src/build/buildpnor/buildpnor.pl#L115) creates the file in default directory, which is [op-build/buildroot](https://github.com/open-power/op-build/blob/e002a0fb45d6bf50ed8d8a270a9110604a8afdfd/op-build-env#L18).

We can set CWD before executing buildpnor.pl, like it does `make` for genPnorImages.pl.
But it looks ugly for me.
May be a function named `load` shouldn't save a file? :)

dcrowell77, 2019-10-14:
See https://github.com/open-power/pnor/pull/130 for a possible fix

artemsen, 2019-11-05:
Thanks!
Works fine in the latest master.

closed 2019-11-05 by artemsen