open-power/hostboot #190: Common PC hardware setups OOM when building Hostboot

stewartsmith, 2019-12-09:
16GB memory, 8 core Ryzen 7 (with HT, so 16 threads) will OOM when building current hostboot, and adjusting parallelism through buildroot configuration does not seem to take effect.

The culprit appears to be somewhere in image generation, with the objdump processes spawned taking up to 2.2GB memory each.

IIRC this is spawned from the Linker, which probably doesn't respect parallelism and instead grabs nprocs or something.

This is all on Fedora 31.

dcrowell77, 2019-12-09:
Seems odd that the OS wouldn't just swap to disk as needed versus failing with OOM.  Do you have some strange virtual memory settings?

stewartsmith, 2019-12-10:
If I add 4GB or swap: still a problem. 12gb in total: it’s okay.

I normally run without swap though.