open-power/hostboot #196: NX Nest AES/SHA accelerators support on PowerNV POWER9 Sforza (and others)

leo-lb, 2021-02-27:
In the POWER9 Sforza processor user manual: https://www-355.ibm.com/systems/power/openpower/posting.xhtml?postingId=FC2A3168C5821FA78525803D00719802

There is:

[image]

Which documents AES and SHA accelerators, however I can't find clues about support in Hostboot, I heard it needs something here.

AES NX accelerators are unsupported on PowerNV in the Linux kernel as well: https://github.com/torvalds/linux/blob/master/drivers/crypto/nx/Kconfig#L3

I wonder, how can this situation be sorted? Since internal documentation to the POWER9 Sforza processor is not available, I don't know what to do to add support here.

Thank you

dcrowell77, 2021-03-01:
Hostboot doesn't take advantage of any of these accelerators.  They are used by code running on top (opal, phyp, linux, whatever).  All of the hardware should be enabled during the boot by default, though there may be some configuration required by the kernel code that ends up using it.

I think a better place for this discussion would be in the opal mailing list.  There should be opal/linux usage somewhere you can use.

closed 2021-03-01 by dcrowell77

klauskiwi, 2021-03-01:
yes, let's move this to the mailing-list. I believe Haren Myneni was working on this, I can make sure that he is aware of the discussion.

leo-lb, 2021-03-02:
@dcrowell77 @klauskiwi 

Thanks, what is the address for that mailing list exactly? Some searches don't show an obvious answer.

I know that Hostboot doesnt use these itself but I don't know from where I heard a while ago there was some support code needed in Hostboot before the accelerator was working, hence why I asked here.

klauskiwi, 2021-03-02:
> 
> 
> @dcrowell77 @klauskiwi
> 
> Thanks, what is the address for that mailing list exactly? Some searches don't show an obvious answer.
> 
https://lists.ozlabs.org/listinfo/openpower-firmware and https://lists.ozlabs.org/listinfo/skiboot are the two main mailing-lists we use