open-power/hostboot #200: In the MCTP protocol stack, is it necessary to enable the link when the BMC is not Ready

Grubby0624, 2021-09-03:
On commit:Iab7b5588ee702c88ccbcb70958912940c0f2f021  file hostlpc.c：
    else
    {
        mctp_prwarn("bmc is not setup");
    }

     // initialize the tx space to say the host is ready to use it
    mctp_binding_set_tx_enabled(&hostlpc->binding, true);

is it necessary to enable the link when the BMC is not Ready?

dcrowell77, 2021-09-05:
@crgeddes ?

crgeddes, 2021-09-07:
No,  we probably should not enable TX if the BMC is reporting that it is **not active**. Really we should be returning a bad RC back out to the caller of mctp_register_bus and triggering a shutdown or retry in this case.

Up until this point, as the BMC is the one who triggers the host to boot, we have assumed that the BMC is ready.

Here is a rough call stack of the MCTP initialization flow as an FYI : 

```
First, the module gets loaded

96  #if defined(CONFIG_MCTP) || defined(CONFIG_MCTP_TESTCASES)
97          /**
98           * @brief   libmctp
99           */
100          {
101                  "libmctp.so" ,                      // taskname
102                  NULL,                               // no pointer to fn
103                  {
104                          START_TASK,
105                          BASE_IMAGE,                  // Base Module
106                  }
107          },
108  #endif

513  // Set the function that will be called when mctp.so is loaded
514  TASK_ENTRY_MACRO( MctpRP::init );

449  void MctpRP::init(errlHndl_t& o_errl)
450  {
451      // This will call the MctpRP construction which initializes MCTP
452      // polling loops
453      return Singleton<MctpRP>::instance()._init();;
454  }
455  

474  void MctpRP::_init(void)
475  {
476      TRACFCOMP(g_trac_mctp, "MctpRP::_init entry");
477  
478      msg_q_register(iv_inboundMsgQ, VFS_ROOT_MSG_MCTP_IN);
479      msg_q_register(iv_outboundMsgQ, VFS_ROOT_MSG_MCTP_OUT);
480  
481      // Get the virtual address for the LPC bar and add the offsets
482      // to the MCTP/PLDM space within  the FW Space of the LPC window.
483      auto l_bar = LPC::get_lpc_virtual_bar() +
484                          LPC::LPCHC_FW_SPACE +
485                          LPC::LPCHC_MCTP_PLDM_BASE;
486  
487      // Initialize the host-lpc binding for hostboot
488      iv_hostlpc = mctp_hostlpc_init_hostboot(l_bar);
489  
490      // Start cmd daemon first because we want it ready if poll daemon finds
491      // something right away
492      task_create(handle_outbound_messages_task, NULL);
493  
494      // Set the receive function to be rx_message which
495      // will handle the message in the RX space accordingly
496      mctp_set_rx_all(iv_mctp, rx_message, NULL);
497  
498      TRACFCOMP(g_trac_mctp, "MctpRP::_init exit");
499      return;
500  }

249  
250  // Public interface used to trigger the initialization of the hostlpc
251  // mctp binding
252  struct mctp_binding_hostlpc *mctp_hostlpc_init_hostboot(uint64_t i_mctpVaddr)
253  {
254    struct mctp_binding_hostlpc *hostlpc;
255  
256    hostlpc = __mctp_hostlpc_init();
[... 284 lines cut ...]
424          iv_mctpVersion = iv_hostlpc->lpc_hdr->negotiated_ver;
425          TRACFCOMP(g_trac_mctp,
426                    "_mctp_channel_init: Negotiated version is : %d",
427                    iv_mctpVersion);
428  
429          if(iv_hostlpc->lpc_hdr->negotiated_ver >= 2)
430          {
431              TRACFCOMP(g_trac_mctp,
432                    "_mctp_channel_init: Setting packet size to be %ld",
433                        iv_hostlpc->lpc_hdr->rx_size );
434              iv_hostlpc->binding.pkt_size = iv_hostlpc->lpc_hdr->rx_size;
435          }
436          printk("MCTP Version : %d    packet size : %d \n",
437                 iv_hostlpc->lpc_hdr->negotiated_ver,
438                 iv_hostlpc->binding.pkt_size);
439  
440      }
441  
442      //else if none of the conditions above are met, the DUMMY_COMMAND is a no op
443  
444  }while(0);
445  
446    return l_errl;
447  }
```

Grubby0624, 2021-12-30:
OK, got it, thanks!

closed 2021-12-30 by Grubby0624