open-power/hostboot #203: How is this ratio determined？

Grubby0624, 2022-01-20:
https://github.com/open-power/hostboot/blob/cb943f25a60c6a4b0409c95b7da12e409e648ab9/src/import/chips/p9/procedures/hwp/memory/lib/workarounds/freq_workarounds.C#L61
Excuse me, how is this ratio determined, and is this the upper limit of nest freq and DIMM freq, so if the nest freq of P9 is 1866MHz, the maximum frequency of DIMM can only reach 2666MHz?

dcrowell77, 2022-01-20:
I believe this is just an architectural limit in the processor memory logic.  Maybe @esteban012 or @stermole can comment more.

stermole, 2022-01-20:
Yes, it's an architectural limit of the logic.  You are correct that the ratio is DIMM freq / nest freq and 1.5 is the upper limit of that ratio.

Grubby0624, 2022-01-24:
Ok, I see. Thank you so much

closed 2022-01-24 by Grubby0624