open-power/hostboot #204: unique_ptr fails to build with GCC 9 (and 10)

shenki, 2022-03-16:
When building hostboot with a newer GCC (see https://github.com/open-power/op-build/pull/4595):
```
    ../../../../src/include/util/impl/unique_ptr.H:408:30: error: cannot
    bind rvalue reference of type ‘std::enable_if<true, STDUniquePtrTest::practice_struct&&>::type’
    {aka ‘STDUniquePtrTest::practice_struct&&’} to lvalue of type ‘STDUniquePtrTest::practice_struct’
      408 |                 return iv_ptr[i];
```

The failure is in the test case (no in hostboot seems to trigger it) here https://github.com/open-power/hostboot/blob/master-p10/src/usr/testcore/lib/unique_ptr.H#L116

```
            i[4].x = 99;

            if (i[4].x != 99)
            {
                TS_FAIL("Array access through unique_ptr returned incorrect value (expected 99)");
            }

            if (i[3].x != 4)
            {
                TS_FAIL("Array access through unique_ptr returned incorrect value (expected 4)");
            }
```

Removing this code allows the build to complete. However, this code seems to be valid. If we build it using GCC's unique_ptr.h, it compiles and the test case runs.

I don't understand what the templates are doing here, but it seems they are the cause of the problem:
```
            /* @brief Index operator for unique_ptr<T[]>.
             *
             * operator[] is not callable on a unique_ptr to non-array types.
             *
             * @param[in] i   Index into the array
             * @return        Value of T::operator[](i)
             */
            template<typename Index, typename X = T>
            typename enable_if<is_array<X>::value, decltype(declval<X>()[declval<Index>()])>::type
            operator[](const Index i)
            {
                return iv_ptr[i];
            }
```

dcrowell77, 2022-03-16:
We've got some time scheduled later this year to bump up our compiler.  In the meantime I'll see if someone can take a quick peek.

ibmzach, 2022-03-16:
Looks like the way that decltype works has changed perhaps. Anyhow, fixed the issue with a commit that should make its way out of gerrit soon.

shenki, 2022-03-17:
Thanks @ibmzach

commit ac366f7d (2022-03-22): Fix unique_ptr<T>::operator[] for GCC 9 and 10

This commit makes a small change to fix operator[] for the unique_ptr
class for GCC 9 and 10. On GCC 9 and 10, operator[] was not returning
a reference and our testcases were failing to compile, even though
GCC 8 does compile it to return a reference.

Also adds a testcase for const unique_ptr and unique_ptr-to-const.

Resolves #204
Change-Id: Ia8bca57fd70656446eb81f8bcca4122cb4d280ff
Reviewed-on: http://rchgit01.rchland.ibm.com/gerrit1/128882
Tested-by: Jenkins OP Build CI <op-jenkins+hostboot@us.ibm.com>
Tested-by: Jenkins Server <pfd-jenkins+h [... cut]