open-power/hostboot #205: GCC 10 can't compile __isSimicsRunning

shenki, 2022-03-17:
This is some funky code that doesn't build:

```
utilmisc.C:31:6: error: ‘bool Util::isSimics()’ specifies less restrictive attribute than its target ‘bool Util::__isSimicsRunning()’: ‘nothrow’ [-Werror=missing-attributes]
   31 | bool isSimics() __attribute__((alias("__isSimicsRunning")));
      |      ^~~~~~~~
utilmisc.C:34:6: note: ‘bool Util::isSimics()’ target declared here
   34 | bool __isSimicsRunning()
      |      ^~~~~~~~~~~~~~~~~
utilmisc.C:48:6: error: ‘bool Util::isQmeModelEnabled()’ specifies less restrictive attribute than its target ‘bool Util::__isQmeEnabled()’: ‘nothrow’ [-Werror=missing-attributes]
   48 | bool isQmeModelEnabled() __attribute__((alias("__isQmeEnabled")));
      |      ^~~~~~~~~~~~~~~~~
utilmisc.C:51:6: note: ‘bool Util::isQmeModelEnabled()’ target declared here
   51 | bool __isQmeEnabled()
      |      ^~~~~~~~~~~~~~
```

shenki, 2022-03-17:
A proposed fix. Note that __isMultiprocSupported doesn't follow the pattern of the first two.
```
diff --git a/src/lib/utilmisc.C b/src/lib/utilmisc.C
index b3ca1ac04c53..85665c824487 100644
--- a/src/lib/utilmisc.C
+++ b/src/lib/utilmisc.C
@@ -28,8 +28,7 @@
 namespace Util
 {
 
-bool isSimics() __attribute__((alias("__isSimicsRunning")));
-extern "C" bool __isSimicsRunning() NEVER_INLINE;
+bool __isSimicsRunning() NEVER_INLINE;
 
 bool __isSimicsRunning()
 {
@@ -40,12 +39,11 @@ bool __isSimicsRunning()
 
 bool isSimicsRunning()
 {
-    static bool simics = isSimics();
+    static bool simics = __isSimicsRunning();
     return simics;
 }
 
 
-bool isQmeModelEnabled() __attribute__((alias("__isQmeEnabled")));
 extern "C" bool __isQmeEnabled() NEVER_INLINE;
 
 bool __isQmeEnabled()
@@ -58,7 +56,7 @@ bool __isQmeEnabled()
 bool requiresSecondaryCoreWorkaround()
 {
     static const auto required =
-        isSimicsRunning() && !isQmeModelEnabled();
+        isSimicsRunning() && !__isQmeEnabled();
     return required;
 }
 
@@ -86,10 +84,14 @@ void setIsConsoleStarted()
     g_isConsoleStarted = true;
 }
 
-bool isMultiprocSupported() __attribute__((alias("__isMultiprocSupported")));
 extern "C" bool __isMultiprocSupported() NEVER_INLINE;
 
-bool __isMultiprocSupported()
+bool __isMulitprocSupported()
+{
+       return MAGIC_INST_CHECK_FEATURE(MAGIC_FEATURE__MULTIPROC);
+}
+
+bool isMultiprocSupported()
 {
     bool multiprocSupport = true;
 
@@ -98,7 +100,7 @@ bool __isMultiprocSupported()
 #else
     if (isSimicsRunning())
     {
-        multiprocSupport = MAGIC_INST_CHECK_FEATURE(MAGIC_FEATURE__MULTIPROC);
+        multiprocSupport = __isMulitprocSupported();
     }
 #endif
```

dcrowell77, 2022-03-17:
Similar to the gcc9 issue, we plan on digging into an upgraded compiler later this year most likely.  Any improvements will be done as time allows.  I'd probably pull in a patch if someone threw one up.

Also, which branch are you attempting this on, master or master-p10?