open-power/hostboot #208: Is the DDR MR setting done by OCMB?

Grubby0624, 2022-07-18:
Excuse me:
The function in the following file src/import/chips/ocmb/explorer/procedures/hwp/memory/lib/exp_getecid_utils.C: read_and_display_normal_training_response，Does it mean that MRS commands are sent to DRAM by OCMB?

dcrowell77, 2022-07-18:
Yes, the OCMB is in charge of all of the DDR communication out to the DRAMs.

@esteban012 @stermole

esteban012, 2022-07-18:
@Grubby0624 
Dan is correct. The OCMB will send the MRS commands to the DRAM.  This is handled in draminit in two sections:

1. The initialization is handled by the OCMB's on chip firmware. The binary package for Explorer is located here https://github.com/open-power/ocmb-explorer-fw
2. After training, some MRS settings are changed via the Configure Command Sequence (CCS).

Hope this helps!

Grubby0624, 2022-07-19:
ok, got it, thanks

closed 2022-07-19 by Grubby0624