open-power/hostboot #210: Read attribute(ATTR_PROC_FAVOR_AGGRESSIVE_PREFETCH) failed at istep 14.8

lili-lilili, 2022-08-11:
We build a pnor and run it on the rainier，hostboot shutdown at istep 14.8

**Here is the console log:**
--== Welcome to SBE - CommitId[0xa9dd6dee] ==--
istep 3.20
istep 3.21
istep 3.22
istep 3.23
istep 4.1
istep 4.2
istep 4.3
istep 4.4
istep 4.5
istep 4.6
istep 4.7
istep 4.8
istep 4.9
istep 4.10
istep 4.11
istep 4.12
istep 4.13
istep 4.14
istep 4.15
istep 4.16
istep 4.17
istep 4.18
istep 4.19
istep 4.20
istep 5.1
istep 5.2
istep 5.3
SBE starting hostboot
istep6.1
Loading boot firmware
Invoking boot firmware

--== Welcome to Hostboot hostboot-p10-7e966f4/hbicore.bin Boot Status (VUART1) ==--

--== FW a5f8d2a a5f8d2a ==--

  1.12541|secure|SecureROM valid - enabling functionality
  1.14390|Current time (Y/M/D H:M:S) : 1970/01/01 00:05:42 @TB=585676161
  1.59679|secure|Booting in non-secure mode (minimum secure version=0x00)
  6.37387|SMT=8, Fuse=0

  7.58207|Booting from Temp side.
  7.60953|Booting from SBE side 0 on master proc=00050000
  7.62071|1970/01/01 00:05:48|ISTEP  6. 4 - host_init_fsi
 13.31740|1970/01/01 00:05:54|ISTEP  6. 5 - host_set_ipl_parms
 13.36366|1970/01/01 00:05:54|ISTEP  6. 6 - host_discover_targets
 19.78645|HWAS|---------------------------------
 19.78645|HWAS|PRESENT>
 19.78842|TARG|PROCS=F0000000
 19.78843|TARG|PROC[00]:
 19.78843|TARG|      CORE=CC0CCCFC DIMM=8800000000000000
 19.78848|TARG|     CACHE=CC0CCCFC OCMB=FF00
 19.78849|TARG|PROC[01]:
 19.78849|TARG|      CORE=00000000 DIMM=0000000000000000
 19.78850|TARG|     CACHE=00000000 OCMB=0000
 19.78851|TARG|PROC[02]:
 19.78851|TARG|      CORE=CF30C33C DIMM=8800000000000000
 19.78852|TARG|     CACHE=CF30C33C OCMB=FF00
 19.78853|TARG|PROC[03]:
 19.78853|TARG|      CORE=00000000 DIMM=0000000000000000
 19.78854|TARG|     CACHE=00000000 OCMB=0000
 19.78855|HWAS|---------------------------------
 19.78881|devtree|Syncing to BMC
 23.97040|1970/01/01 00:06:04|ISTEP  6. 7 - host_update_primary_tpm
 23.99098|1970/01/01 00:06:04|ISTEP  6. 8 - host_gard
 24.06329|HWAS|---------------------------------
 24.06330|HWAS|FUNCTIONAL>
 24.06453|TARG|PROCS=F0000000
 24.06454|TARG|PROC[00]:
 24.06454|TARG|      CORE=CC0CCCFC DIMM=8800000000000000
 24.06455|TARG|     CACHE=CC0CCCFC OCMB=FF00
 24.06456|TARG|PROC[01]:
 24.06456|TARG|      CORE=00000000 DIMM=0000000000000000
 24.06457|TARG|     CACHE=00000000 OCMB=0000
 24.06458|TARG|PROC[02]:
 24.06458|TARG|      CORE=CF30C33C DIMM=8800000000000000
 24.06459|TARG|     CACHE=CF30C33C OCMB=FF00
 24.06460|TARG|PROC[03]:
 24.06460|TARG|      CORE=00000000 DIMM=0000000000000000
 24.06461|TARG|     CACHE=00000000 OCMB=0000
 24.06462|HWAS|---------------------------------
 24.71534|SECURE|Security Access Bit> 0x0000000000000000
 24.71534|SECURE|Secure Mode Disable (via Jumper)> 0xF000000000000000
 24.95477|1970/01/01 00:06:05|ISTEP  6. 9 - host_voltage_config
 25.96584|1970/01/01 00:06:06|ISTEP  7. 1 - mss_attr_cleanup
 27.01120|1970/01/01 00:06:07|ISTEP  7. 2 - mss_volt
 27.19791|1970/01/01 00:06:08|ISTEP  7. 3 - mss_freq
 27.62807|1970/01/01 00:06:08|ISTEP  7. 4 - mss_eff_config
 28.30535|1970/01/01 00:06:09|ISTEP  7. 5 - mss_attr_update
 28.36949|1970/01/01 00:06:09|ISTEP  8. 1 - host_setup_sbe
 28.59018|1970/01/01 00:06:09|ISTEP  8. 2 - host_secondary_sbe_config
 28.71655|1970/01/01 00:06:09|ISTEP  8. 3 - host_cbs_start
 28.76719|1970/01/01 00:06:09|ISTEP  8. 4 - proc_check_secondary_sbe_seeprom_complete
 44.02630|1970/01/01 00:06:24|ISTEP  8. 5 - host_attnlisten_proc
 44.16484|1970/01/01 00:06:25|ISTEP  8. 6 - proc_fbc_eff_config
 44.24190|1970/01/01 00:06:25|ISTEP  8. 7 - proc_eff_config_links
 44.32879|1970/01/01 00:06:25|ISTEP  8. 8 - proc_attr_update
 44.39696|1970/01/01 00:06:25|ISTEP  8. 9 - proc_chiplet_fabric_scominit
 45.62604|1970/01/01 00:06:26|ISTEP  8.10 - host_set_voltages
 45.83907|1970/01/01 00:06:26|ISTEP  8.11 - proc_io_scominit
 46.96895|1970/01/01 00:06:27|ISTEP  8.12 - proc_load_ioppe
 49.55555|1970/01/01 00:06:30|ISTEP  8.13 - proc_iohs_enable_ridi
 49.60738|1970/01/01 00:06:30|ISTEP  8.14 - proc_init_ioppe
 53.56197|1970/01/01 00:06:34|ISTEP  9. 1 - proc_io_dccal_done
 53.87013|1970/01/01 00:06:34|ISTEP  9. 2 - fabric_dl_pre_trainadv
 53.88946|1970/01/01 00:06:34|ISTEP  9. 3 - fabric_dl_setup_training
 53.93503|1970/01/01 00:06:34|ISTEP  9. 4 - proc_fabric_link_layer
 54.00329|1970/01/01 00:06:34|ISTEP  9. 5 - fabric_dl_post_trainadv
 54.02488|1970/01/01 00:06:34|ISTEP  9. 6 - proc_fabric_iovalid
 54.30395|1970/01/01 00:06:35|ISTEP  9. 7 - proc_fbc_eff_config_aggregate
 54.34287|1970/01/01 00:06:35|ISTEP 10. 1 - proc_build_smp
 55.79832|1970/01/01 00:06:36|ISTEP 10. 2 - host_sbe_update
 63.22325|sbe|System Performing SBE Update for PROC 3, side 1
 63.22333|sbe|System Performing SBE Update for PROC 0, side 1
 63.22335|sbe|System Performing SBE Update for PROC 2, side 1
 63.22337|sbe|System Performing SBE Update for PROC 1, side 1
157.31192|1970/01/01 00:08:18|ISTEP 10. 3 - host_secureboot_lockdown
161.11255|1970/01/01 00:08:21|ISTEP 10. 4 - proc_chiplet_scominit
161.74637|1970/01/01 00:08:22|ISTEP 10. 5 - proc_pau_scominit
161.84839|1970/01/01 00:08:22|ISTEP 10. 6 - proc_pcie_scominit
161.98813|1970/01/01 00:08:22|ISTEP 10. 7 - proc_scomoverride_chiplets
162.01978|1970/01/01 00:08:22|ISTEP 10. 9 - host_rng_bist
162.04578|1970/01/01 00:08:22|ISTEP 11. 1 - host_prd_hwreconfig
163.28373|1970/01/01 00:08:24|ISTEP 11. 2 - host_set_mem_volt
164.15929|1970/01/01 00:08:25|ISTEP 11. 3 - proc_ocmb_enable
166.30930|1970/01/01 00:08:27|ISTEP 11. 4 - ocmb_check_for_ready
176.84151|1970/01/01 00:08:37|ISTEP 12. 1 - mss_getecid
177.66286|1970/01/01 00:08:38|ISTEP 12. 2 - omi_attr_update
177.68424|1970/01/01 00:08:38|ISTEP 12. 3 - proc_omi_scominit
177.80148|1970/01/01 00:08:38|ISTEP 12. 4 - ocmb_omi_scominit
177.82554|1970/01/01 00:08:38|ISTEP 12. 5 - omi_pre_trainadv
178.02092|1970/01/01 00:08:38|ISTEP 12. 6 - omi_setup
187.08408|1970/01/01 00:08:47|ISTEP 12. 7 - omi_io_run_training
187.96768|1970/01/01 00:08:48|ISTEP 12. 8 - omi_train_check
188.69434|1970/01/01 00:08:49|ISTEP 12. 9 - omi_post_trainadv
189.25617|1970/01/01 00:08:50|ISTEP 12.10 - host_attnlisten_memb
189.46883|1970/01/01 00:08:50|ISTEP 12.11 - host_omi_init
190.26814|1970/01/01 00:08:51|ISTEP 12.12 - update_omi_firmware
190.99071|1970/01/01 00:08:51|ISTEP 13. 1 - mss_scominit
191.68042|1970/01/01 00:08:52|ISTEP 13. 2 - mss_draminit
192.24401|1970/01/01 00:08:53|ISTEP 13. 3 - mss_draminit_mc
193.37218|1970/01/01 00:08:54|ISTEP 14. 1 - mss_memdiag
196.86365|1970/01/01 00:08:57|ISTEP 14. 2 - mss_thermal_init
197.64199|1970/01/01 00:08:58|ISTEP 14. 3 - proc_load_iop_xram
198.24088|1970/01/01 00:08:59|ISTEP 14. 4 - proc_pcie_config
198.31376|1970/01/01 00:08:59|ISTEP 14. 5 - proc_setup_mmio_bars
198.35547|1970/01/01 00:08:59|ISTEP 14. 6 - host_secure_rng
198.39247|1970/01/01 00:08:59|ISTEP 14. 7 - host_enable_memory_encryption
198.71736|1970/01/01 00:08:59|ISTEP 14. 8 - proc_exit_cache_contained
199.04684|================================================
199.06696|Error reported by fapi2 (0x3300) EID 0x90000A5B
199.08138|  Invalid attribute read request
199.08139|  ModuleId   0x0c fapi2::MOD_FAPI2_GET_TARGETING_ATTR
199.08139|  ReasonCode 0x332a RC_INVALID_ATTRIBUTE
199.08139|  UserData1  FAPI2 Target Type : 0xffffffff00010000
199.09749|  UserData2  HB Target HUID : 0x00000000055ddfeb
199.09749|------------------------------------------------
199.09750|  p10_exit_cache_contained
199.09750|------------------------------------------------
199.09750|  Callout type               : Procedure Callout
199.09751|  Procedure                  : EPUB_PRC_HB_CODE
199.09751|  Priority                   : SRCI_PRIORITY_LOW
199.09751|------------------------------------------------
199.09752|  Hostboot Build ID: hostboot-p10-7e966f4/hbicore.bin
199.09752|================================================
199.47717|System shutting down with error status 0x90000A5B
200.04092|================================================
200.04092|Error reported by istep (0x1700) EID 0x90000A5C
200.06103|  IStep failed, Check other log(s) with the same PLID for reason.
200.06103|  ModuleId   0x01 MOD_REPORTING_ERROR
200.06104|  ReasonCode 0x1703 RC_FAILURE
200.06104|  UserData1  eid of first error : 0x90000a5b0000332a
200.06104|  UserData2  Reason code of first error : 0x0000000100000e08
200.06105|------------------------------------------------
200.06105|  proc_exit_cache_contained
200.06105|------------------------------------------------
200.06106|  Callout type               : Procedure Callout
200.06106|  Procedure                  : EPUB_PRC_HB_CODE
200.06106|  Priority                   : SRCI_PRIORITY_LOW
200.06107|------------------------------------------------
200.06107|  Hostboot Build ID: hostboot-p10-7e966f4/hbicore.bin
200.06107|================================================
200.13360|devtree|Syncing to BMC

lili-lilili, 2022-08-11:
**Here is the log with trace:**

210.15929|ATTN_FAST|I>...intr task woke up
210.15932|INTR|IntrRp::msgHandler() lsiIntStatus 0x0000000000000000
210.15933|INTR|IntrRp::msgHandler(): LSI Interrupt Detected
210.15934|INTR|IntrRp::msgHandler() lsiIntStatus 0x0000000000000000
210.15934|INTR|IntrRp::msgHandler() lsiIntStatus 0x0000000000000000
210.15935|INTR|IntrRp::msgHandler() lsiIntStatus 0x0000000000000000
210.15936|INTR|IntrRp::msgHandler() lsiIntStatus 0x0000000000000000
210.15938|INTR|IntrRp::completeInterruptProcessing() Removing pending interrupt for pir: 0x0,interrupt type: 4
210.15992|PRDF|I>PRD Signature 00050000 34EB0000
210.15993|PRDF|I>PRD Signature 00050000 34EB000D
210.15995|PRDF|I>PRD Signature 00050000 34EB000D
210.16033|INTR|IntrRp::completeInterruptProcessing() Removing pending interrupt for pir: 0x400,interrupt type: 4
210.16037|PRDF|I>PRD Signature 00440001 00000000
210.16067|PRDF|I>PRD Signature 00440001 65CC0000
210.16069|PRDF|I>PRD Signature 00440001 65CC0007
210.16070|PRDF|I>PRD Signature 00440001 65CC0007
210.16074|PRDF|I>PRD Signature 00490003 00000000
210.16096|PRDF|I>PRD Signature 00490003 E6E60000
210.16113|PRDF|I>PRD Signature 00490003 E6E60000
210.16114|PRDF|I>PRD Signature 00490003 E6E60000
210.16124|PRDF|I>PRD Signature 00490003 F0C60000
210.16127|PRDF|I>PRD Signature 00490003 F0C60006
210.16129|PRDF|I>PRD Signature 00490003 F0C60006
210.16135|PRDF|I>PRD Signature 004B0007 00000000
210.16153|PRDF|I>PRD Signature 004B0007 1CD40000
210.16155|PRDF|I>PRD Signature 004B0007 1CD40005
210.16156|PRDF|I>PRD Signature 004B0007 1CD40005
210.16252|PRDF|I>PRD Signature 004B0007 59930000
210.16255|PRDF|I>PRD Signature 004B0007 5993000A
210.16256|PRDF|I>PRD Signature 004B0007 5993000A
210.16257|MDIA_FAST|I>sm: processMaintCommandEvent
210.16258|MDIA_FAST|I>sm: processing event for: 0x004b0007, target: 0x004b0007, type: 2
210.16259|PLDM|I> >>resetWatchdogTimer
210.18625|PLDM|I> <<resetWatchdogTimer
210.18640|INITSVC|sendProgressCodeToBmc(): sending CC009381 progress code
210.19038|PRDF|[MemTdCtlr::defaultStep] The TD command made it to the end of memory on chip: 0x004b0007
210.19047|MDIA_FAST|I>sm: processMaintCommandEvent
210.19048|MDIA_FAST|I>sm: processing event for: 0x004b0007, target: 0x004b0007, type: 0
210.19049|PLDM|I> >>resetWatchdogTimer
210.19860|PLDM|I> <<resetWatchdogTimer
210.19873|INITSVC|sendProgressCodeToBmc(): sending CC009381 progress code
210.20384|MDIA_FAST|I>sm: dispatching work item 13 for: 0x004b0007, priority: 2, 
210.20437|MDIA_FAST|I>sm: executing work item 13 for: 0x004b0007
210.20445|MDIA_FAST|I>sm: dispatching work item 15 for: 0x004b0007, priority: 1, 
210.20446|MDIA_FAST|I>sm: executing work item 15 for: 0x004b0007
210.20446|ERRL|E>Error deleted without commit : PLID=90000D89, EID=90000D89
210.20447|MDIA_FAST|I>Running mss::unmask::after_memdiags HWP call on OCMB target HUID 0x004B0007.
210.20450|PRDF|<<PRDF::main()
210.20455|ATTN_SLOW|I>callPrd with Attn Count of 1
210.20456|PRDF|>>PRDF::main() Global attnType=0005
210.20457|INTR|IntrRp::msgHandler(): LSI Interrupt Detected
210.20458|INTR|IntrRp::msgHandler() lsiIntStatus 0x0000000000000000
210.20458|INTR|IntrRp::msgHandler() lsiIntStatus 0x0000000000000000
210.20459|INTR|IntrRp::msgHandler() lsiIntStatus 0x0800000000000000
210.20460|INTR|IntrRp::msgHandler() External Interrupt found for pir: 0x400,interrupt type: 4
210.20461|MDIA_FAST|I>SUCCESS: mss::unmask::after_memdiags HWP call on OCMB target HUID 0x004b0007.
210.20463|MDIA_FAST|I>Running mss::reset_reorder_queue_settings HWP call on OCMB target HUID 0x004B0007.
210.20471|ERRL|E>Error created : PLID=90000D8A, EID=90000D8A, RC=E580, Mod=0B, Userdata=0000000000000000 0000000000000000, Sev=RECOVERED
210.20471|MDIA_FAST|I>SUCCESS: mss::reset_reorder_queue_settings HWP call on OCMB target HUID 0x004b0007.
210.20473|PRDF|I>PRD Signature 00000000 00000000
210.20473|PRDF|I>PRD Signature 00050002 00000000
210.20484|ATTN_FAST|I>...intr task woke up
210.20561|INTR|IntrRp::msgHandler() lsiIntStatus 0x0000000000000000
210.20572|INTR|IntrRp::completeInterruptProcessing() Removing pending interrupt for pir: 0x400,interrupt type: 4
210.20614|PRDF|I>PRD Signature 00050002 34EB0000
210.20615|PRDF|I>PRD Signature 00050002 34EB000D
210.20616|PRDF|I>PRD Signature 00050002 34EB000D
210.20636|PRDF|I>PRD Signature 00440009 00000000
[... 690 lines cut ...]
215.64180|INITSVC|flushing virtual memory 0x9e000000
215.64181|INITSVC|flushing virtual memory 0x9d000000
215.64185|INITSVC|flushing virtual memory 0x9c000000
215.64186|INITSVC|flushing virtual memory 0x9a000000
215.64186|INITSVC|flushing virtual memory 0x99000000
215.64187|INITSVC|flushing virtual memory 0x98000000
215.64187|INITSVC|flushing virtual memory 0x97000000
215.64188|INITSVC|flushing virtual memory 0x93000000
215.64188|INITSVC|flushing virtual memory 0x92000000
215.64189|INITSVC|flushing virtual memory 0x90000000
215.64189|INITSVC|flushing virtual memory 0x8f000000
215.64190|INITSVC|flushing virtual memory 0x8d000000
215.64191|INITSVC|flushing virtual memory 0x8c000000
215.64191|INITSVC|flushing virtual memory 0x89000000
215.64192|INITSVC|flushing virtual memory 0x88000000
215.64192|INITSVC|flushing virtual memory 0x87000000
215.64217|INITSVC|flushing virtual memory 0x86000000
215.64219|INITSVC|flushing virtual memory 0x85000000
215.64219|INITSVC|flushing virtual memory 0x84000000
215.64222|INITSVC|flushing virtual memory 0x83000000
215.64222|INITSVC|flushing virtual memory 0x82000000
215.64223|INITSVC|flushing virtual memory 0x81000000
215.64418|INITSVC|Do post mem flush callbacks
215.64418|PNOR|PnorRP::Shutdown message recieved
215.65420|INITSVC|_doShutdown>

dcrowell77, 2022-08-11:
This kind of error tends to be a problem with patching the system.  Was this a complete flash build or were there code patches involved?

lili-lilili, 2022-08-12:
> This kind of error tends to be a problem with patching the system. Was this a complete flash build or were there code patches involved?

Hi dcrowell：
There is no extended patch.
I did a full compilation with op-build ibm-release-fw1020 branch(https://github.com/open-power/op-build/tree/ibm-release-fw1020)。

dcrowell77, 2022-08-12:
How did you package the op-build output into the BMC image?  That is where I suspect a problem.  The attribute data comes from a lid on the BMC.  If you run `pldmtool bios GetBIOSAttributeCurrentValueByHandle -a hb_lid_ids` and look for the HBD entry it will tell you the specific lid that holds our attributes.

The failing attribute is ATTR_PROC_FAVOR_AGGRESSIVE_PREFETCH (see obj/genfiles/attributeenums.H in the hostboot tree for the decode).  I know that attribute was added pretty recently.  I suspect that you haven't properly integrated the op-build into the BMC image correctly somehow such that your BMC image is using an older version of the data than the Hostboot code expects.  (That part of the flow is not my area of expertise.)

lili-lilili, 2022-08-17:
> How did you package the op-build output into the BMC image? That is where I suspect a problem. The attribute data comes from a lid on the BMC. If you run `pldmtool bios [...]

Thanks a lot, this is indeed a problem caused by lid mapping。

closed 2022-08-17 by lili-lilili