open-power/hostboot #211: How to read the SCOM of OCMB

Grubby0624, 2022-10-20:
Here's a question:
https://github.com/open-power/hostboot/blob/44a70705c9725087850e7bf466375d7f5147e9c0/src/import/chips/ocmb/explorer/common/include/explorer_scom_addresses_fixes.H#L33

1. As for the SCOM register above, I understand that this register represents RX CLK DLY of P0 U0 TG0 DBYTE0, but shouldn't RX CLK DLY be a "per nibble" value?

2. What does "P0" mean by pstate0 and "U0" mean?

3. If I want to get the value of this register, I should go through getscom (ocmb_target, address). Is this the case?

dcrowell77, 2022-10-20:
I can answer 3 easily - Yes.

The other questions are a bit outside my scope as they are internal to the Explorer OCMB Chip hardware logic.  I'll dig in a bit to see if I can find any better documentation from Microchip.

dcrowell77, 2022-10-20:
Here's the info from an internal register document.  This is all hardware-specifics that I can't speak to.

Register 0x4040230: DDR4_PHY:PHY - DBYTE0_RXCLKDLYTG0_U0_P0
Trained Read DQS to RxClk Delay (Timing Group DEST=0).
Bit, Type, Function, Default
15:6, -, UNUSED, XX XXXX XXXX
5:0, R/W, RXCLKDLYTG0_U0_P0, 01 0000

RXCLKDLYTG0_U0_P0
Trained Read DQS to RxClk Delay (Timing Group DEST=0).
Trained to generate read DQ receive clock from the received read DQS.
RxClk Delay (6 bits, fine delay only); (For Timing Group 0)
RxClkDlyTg0[4:0] is the fine delay, ie one unit of delay is one-thirtysecond of a UI = UI/32.
RxClkDlyTg0[5] is reserved. There is no coarse delay field, ie the max configurable delay is 31/32 UI.
Register Block Offset Address 0x08c controls the lower DQ nibble RxClk timing. (For Timing Group 0)
Register Block Offset Address 0x18c controls the upper DQ nibble RxClk timing. (For Timing Group 0)
These Registers are replicated per pstate.
.
Note: Timing Group 0 = RANK0.

dcrowell77, 2022-10-20:
There might be more hints in the Explorer Firmware out at https://github.com/open-power/ocmb-explorer-fw/ too.  Good luck!

closed 2022-10-20 by dcrowell77

Grubby0624, 2022-10-21:
Okay, thank you very much