open-power/hostboot #219: Hostboot start failed

lili-lilili, 2023-02-28:
We are using OpenBMC to boot Rainier.
It seems that SBE and HBBL can work well, but we can't see hostboot welcome log “--== Welcome to Hostboot xxxx ==--” 
We want to see the printk log of hostboot.
Can anyone provide a method to dump printk logs?

--== Welcome to SBE - CommitId[0x2f004223] ==--

istep 3.20

istep 3.21

istep 3.22

istep 3.23

istep 4.1

istep 4.2

istep 4.3

istep 4.4

istep 4.5

istep 4.6

istep 4.7

istep 4.8

istep 4.9

istep 4.10

istep 4.11

istep 4.12

istep 4.13

istep 4.14

istep 4.15

istep 4.16

istep 4.17

istep 4.18

istep 4.19

istep 4.20

istep 5.1

istep 5.2

istep 5.3

SBE starting hostboot

istep6.1
Loading boot firmware
Invoking boot firmware

dcrowell77, 2023-02-28:
The BMC code should have generated a hostboot dump that contains the memory contents.  The memory has to be extracted from the SYSDUMP file by using the makedump tool.  Alternatively if security is disabled you can pull the memory directly using https://github.com/open-power/hostboot/blob/master-p10/src/build/debug/hb-memdump.sh .

Once you have the memory contents you can use https://github.com/open-power/hostboot/blob/master-p10/src/build/debug/hb-dump-debug to execute any of the methods of our debug framework.  The various modules are listed under https://github.com/open-power/hostboot/tree/master-p10/src/build/debug/Hostboot .  The most useful tend to be Printk, Trace, and Ps.

For printk specifically you would run this from the root of the hostboot repo that built the image you are running.
`hb-dump-debug --file=hbmem.bin --tool=Printk`

lili-lilili, 2023-03-01:
Thanks，it's very helpful

closed 2023-03-01 by dcrowell77