open-power/hostboot #224: Power10 small core cpu checkstop

lili-lilili, 2023-05-17:
When we test a Power10 small core CPU on the rainier, CPU will checkstop due to "L3 Dir read UE".
But if we use a Big core CPU, there is no problem.
Is there anything to need special attention when using small core cpu? Like some attribute settings, or some config? 

**Here is the event log:**
{
"Private Header": {
    "Section Version":          "1",
    "Sub-section type":         "0",
    "Created by":               "0xE500",
    "Created at":               "05/17/2023 05:48:45",
    "Committed at":             "05/17/2023 05:48:45",
    "Creator Subsystem":        "BMC",
    "CSSVER":                   "",
    "Platform Log Id":          "0x50000011",
    "Entry Id":                 "0x50000011",
    "BMC Event Log Id":         "243"
},
"User Header": {
    "Section Version":          "1",
    "Sub-section type":         "0",
    "Log Committed by":         "0x2000",
    "Subsystem":                "Processor Unit (CPU)",
    "Event Scope":              "Entire Platform",
    "Event Severity":           "Unrecoverable Error",
    "Event Type":               "Not Applicable",
    "Action Flags": [
                                "Service Action Required",
                                "Report Externally",
                                "HMC Call Home"
    ],
    "Host Transmission":        "Not Sent",
    "HMC Transmission":         "Not Sent"
},
"Primary SRC": {
    "Section Version":          "1",
    "Sub-section type":         "1",
    "Created by":               "0xE500",
    "SRC Version":              "0x02",
    "SRC Format":               "0x55",
    "Virtual Progress SRC":     "False",
    "I5/OS Service Event Bit":  "False",
    "Hypervisor Dump Initiated":"False",
    "Backplane CCIN":           "2E2F",
    "Terminate FW Error":       "False",
    "Deconfigured":             "False",
    "Guarded":                  "True",
    "Error Details": {
        "Message":              "Error Signature: 0x20DA0020 0x00020001 0x5074140E"
    },
    "Valid Word Count":         "0x09",
    "Reference Code":           "BD13E510",
    "Hex Word 2":               "00000055",
    "Hex Word 3":               "2E2F0010",
    "Hex Word 4":               "00000000",
    "Hex Word 5":               "01000000",
    "Hex Word 6":               "20DA0020",
    "Hex Word 7":               "00020001",
    "Hex Word 8":               "5074140E",
    "Hex Word 9":               "00000000",
    "Callout Section": {
        "Callout Count":        "1",
        "Callouts": [{
            "FRU Type":         "Normal Hardware FRU",
            "Priority":         "Medium Priority",
            "Location Code":    "U78DB.ND0.WZS008L-P0-C24",
            "Part Number":      "F210110",
            "CCIN":             "AB42",
            "Serial Number":    "            "
        }]
[... 264 lines cut ...]
        "node 0 ocmb 32 (Explorer 2.0) ******************************",
        "  CHIPLET_OCMB_FIR_MASK     (0x08040002) 6627 FFE0 0000 0000",
        "node 0 ocmb 34 (Explorer 2.0) ******************************",
        "  CHIPLET_OCMB_FIR_MASK     (0x08040002) 6627 FFE0 0000 0000",
        "node 0 ocmb 36 (Explorer 2.0) ******************************",
        "  CHIPLET_OCMB_FIR_MASK     (0x08040002) 6627 FFE0 0000 0000",
        "node 0 ocmb 38 (Explorer 2.0) ******************************",
        "  CHIPLET_OCMB_FIR_MASK     (0x08040002) 6627 FFE0 0000 0000",
        "node 0 proc 3 (P10 2.0) ************************************",
        "  GFIR_CS                   (0x570F001C) 1000 0000 0000 0000",
        "  CFIR_N1_CS                (0x03040000) 8000 0000 4000 0000",
        "  CFIR_N1_CS_MASK           (0x03040040) 2000 0000 0000 0000",
        "  PB_EXT_FIR                (0x030113AE) 4000 0000 0000 0000",
        "  PB_EXT_FIR_MASK           (0x030113B1) B400 0000 0000 0000",
        "node 0 ocmb 56 (Explorer 2.0) ******************************",
        "  CHIPLET_OCMB_FIR_MASK     (0x08040002) 6627 FFE0 0000 0000",
        "node 0 ocmb 58 (Explorer 2.0) ******************************",
        "  CHIPLET_OCMB_FIR_MASK     (0x08040002) 6627 FFE0 0000 0000",
        "node 0 ocmb 60 (Explorer 2.0) ******************************",
        "  CHIPLET_OCMB_FIR_MASK     (0x08040002) 6627 FFE0 0000 0000",
        "node 0 ocmb 62 (Explorer 2.0) ******************************",
        "  CHIPLET_OCMB_FIR_MASK     (0x08040002) 6627 FFE0 0000 0000"
    ]
}
}

dcrowell77, 2023-05-17:
I assume you are booting into OPAL?  Do you have all power management disabled?  What is the last thing you see in the boot console?

This piece of FFDC points to the IPL being through Hostboot and into the payload (PHYP or OPAL).
"Hostboot Scratch Registers": {
  "0x0000283c": "0xaa811504",     ==> istep 21.4
  "0x000000004602f489": "0x0000000000000000"    ==> Says "hostboot" while we're running, set to zero when we jump to payload
}

You should probably engage the Opal team (https://github.com/open-power/skiboot) to see if they are aware of any issues.

mikey, 2023-05-19:
@lili-lilili Do you have the OPAL boot log for the small core case so we can take a look?

lili-lilili, 2023-05-19:
> @lili-lilili Do you have the OPAL boot log for the small core case so we can take a look?

For OPAL log: https://github.com/open-power/skiboot/issues/275

dcrowell77, 2023-06-26:
This is being discussed in the skiboot repo so closing this issue