open-power/hostboot #228: Is the memory stream performance related to Signal integrity

Grubby0624, 2023-06-25:
https://github.com/open-power/hostboot/blob/abc40c539e2cb3cb5d7c1473ad9122488b3a595c/src/import/chips/ocmb/explorer/procedures/hwp/memory/exp_draminit.C#L62
During the stream test, I found that the measured stream result is less than 50% of the theoretical value. I think that Signal integrity only affects whether memory initialization can be successful. May I ask for your advice:
    Does Signal integrity affect memory stream performance? If I want to debug this problem, do I need to pay attention to the Signal integrity parameters I choose?

dcrowell77, 2023-06-26:
We aren't sure what you mean by "Signal integrity" here.  If you are talking about electrical issues on any of the busses that can definitely impact performance.  However, you mentioned "Signal integrity parameters" which makes me think you are referring to some kind of setting.   Can you provide more detail on what/where you are using these parameters?

Also, where are you measuring your results and how?  OCMB to DDR?  Proc to OCMB?  The complete path?

ecorderoibm, 2023-06-26:
I don't suspect that the signal integrity settings (assuming you're talking about DDR) would cause an impact to the performance. It should, like you say, really get you to be able to initialize. If you have some settings you might suspect, we can take look at those. I would think you would want to look at the MC settings first though. I suspect it be more likely in there.

Grubby0624, 2023-06-28:
1. On our platform, the memory Training cannot pass, so I modified some parameters in the' user_input_msdg 'sent by PNOR to OCMB to pass the memory training:
        PhyOdtImpedance = 80 60 48 40
        PhyDrvImpedancePU = 60 53 48 43 40 36 34 32 30
        PhyDrvImpedancePD = 36 34 32 30 28
        DramRttNomR0/1 = 240 120 80 60 40 34
        DramRttWrR0/1 = 240 120 80
        DramRttParkR0/1 = 240 120 80 60 48 40 34
Some of these parameter combinations can be trained through memory and booted to the OS, such as the following group:
        PhyOdtImpedance = 80
        PhyDrvImpedancePU = 60
        PhyDrvImpedancePD = 36
        DramRttNomR0/1 = 120
        DramRttWrR0/1 = 80
        DramRttParkR0/1 =40
But after entering the OS, we used the "stream" test tool to test and found that the Memory bandwidth was only 25% of the theoretical value. I suspect that the "user_input_msdg" value I selected affected the stream performance.

2. where are you measuring your results and how?
      What we are testing should be the stream performance of the entire link. The following stream testing tools were used:
              https://www.cs.virginia.edu/stream/ref.html
      So is there a way to test the bandwidth performance of OCMB>DIMM separately?

3. When running stream testing under OS, sometimes the following error will be reported and the test will stop. 
                [  769.211099] Disabling lock debugging due to kernel taint
                [  769.211102] MCE: CPU4: machine check (Severe) Host UE Load/Store DAR: 00000000ad56ba80 paddr: 0000000383a30000 [Not recovered]
                [  769.211103] MCE: CPU4: PID: 3970 Comm: stream NIP: [00000000100014c0]
                [  769.211103] MCE: CPU4: Initiator CPU
                [  769.211103] MCE: CPU4: Hardware error
                [  769.211104] stream[3970]: unhandled signal 7 at 00000000100014c0 nip 00000000100014c0 lr 0000000010001448 code 4
                [  769.212179] Memory failure: 0x383a3: recovery action for dirty LRU page: Recovered
    May I ask:
          Is this also related to the parameter values I selected.
4 .Finally, we currently choose the ' user_input_msdg ' value based on the condition of being able to train in memory and achieve optimal stream performance. Is there a better way to determine which "user_input_msdg" parameter value is optimal for our platform?

dcrowell77, 2023-06-28:
> Memory bandwidth was only 25% of the theoretical value. 

How do the numbers compare to an equivalent Rainier configuration?  Comparing to a real-world example is perhaps more accurate than theory.

3. That would imply some settings are wrong and the memory is not running correctly.  @ecorderoibm can probably answer more intelligently.  Do you see any PELs coming from opal-prd in this case?

ecorderoibm, 2023-06-28:
those settings you changed don't have any affect on the bandwidth are only for signal integrity on the DRAM interface. I agree with Dan. Would be good to know what you measured using a DDIMM and work backwards from there for comparison and validation of you measurements.

Grubby0624, 2023-06-29:
1. The following are my test results. Due to the limitations of our platform, I inserted a 3200MHz ISDIMM under the OMI 0A/1A/2A/3A of Proc0. When testing the bandwidth of the DIMM under Proc0, the result was 24558.3MB/s. On Rainier, I inserted a 3200MHz DDIMM under the 0A/1A/2A/3A of Proc0, and the result was 74540.4MB/s.
PS: The theoretical bandwidth of four 3200MHz DIMMs is 100GB/s, which has been tested on Rainier and reaches 65% of the theoretical value
2. When I run the stream script in OS and  report an MCE error, I cannot see any PEL in BMC. However, if I copy the script to ‘/etc/profile. d/’ and execute the script automatically during OS startup, the system will report checkststop. I will separate the collected PEL logs into my later answer.

Grubby0624, 2023-06-29:
This is the PEL log:
{
"Private Header": {
    "Section Version":          "1",
    "Sub-section type":         "0",
    "Created by":               "0xE500",
    "Created at":               "12/21/2022 18:04:46",
    "Committed at":             "12/21/2022 18:04:46",
    "Creator Subsystem":        "BMC",
    "CSSVER":                   "",
    "Platform Log Id":          "0x500006C6",
    "Entry Id":                 "0x500006C6",
    "BMC Event Log Id":         "2449"
},
"User Header": {
    "Section Version":          "1",
    "Sub-section type":         "0",
    "Log Committed by":         "0x2000",
    "Subsystem":                "Memory DIMM",
    "Event Scope":              "Entire Platform",
    "Event Severity":           "Unrecoverable Error",
    "Event Type":               "Not Applicable",
    "Action Flags": [
                                "Service Action Required",
                                "Report Externally",
                                "HMC Call Home"
    ],
    "Host Transmission":        "Not Sent",
    "HMC Transmission":         "Not Sent"
},
"Primary SRC": {
    "Section Version":          "1",
    "Sub-section type":         "1",
    "Created by":               "0xE500",
    "SRC Version":              "0x02",
    "SRC Format":               "0x55",
    "Virtual Progress SRC":     "False",
    "I5/OS Service Event Bit":  "False",
    "Hypervisor Dump Initiated":"False",
    "Backplane CCIN":           "2E2D",
    "Terminate FW Error":       "False",
    "Deconfigured":             "False",
    "Guarded":                  "True",
    "Error Details": {
        "Message":              "Error Signature: 0x60D20020 0x00020003 0xCCAA000E"
    },
    "Valid Word Count":         "0x09",
    "Reference Code":           "BD23E510",
    "Hex Word 2":               "00080055",
    "Hex Word 3":               "2E2D0010",
    "Hex Word 4":               "CC009544",
    "Hex Word 5":               "01000000",
    "Hex Word 6":               "60D20020",
    "Hex Word 7":               "00020003",
    "Hex Word 8":               "CCAA000E",
    "Hex Word 9":               "00000000",
    "Callout Section": {
        "Callout Count":        "1",
        "Callouts": [{
            "FRU Type":         "Normal Hardware FRU",
            "Priority":         "Mandatory, replace all with this type as a unit",
            "Location Code":    "U78DA.ND0.WZS01SL-P0-C27",
            "Part Number":      "",
            "CCIN":             "",
            "Serial Number":    ""
        }]
    },
    "SRC Details": {
        "Primary Attention": "system checkstop",
        "Signature Description": {
[... 446 lines cut ...]
        "  MCBMCAT                   (0x080118D7) 0004 E201 CC00 0000",
        "  MCB_CNTLSTAT              (0x080118DC) 8000 0000 0000 0000",
        "  MCBCFG                    (0x080118E0) 0000 0000 0000 0020",
        "  EXP_MSR                   (0x08011C0C) 00CA 2000 0000 0000",
        "  RDF_ERR_RPT_1             (0x08011C0F) 0000 3000 0000 0000",
        "  HW_MS                     (0x08011C10) CAC0 0000 0000 0000",
        "  HW_MS                     (0x08011C11) CAC0 0000 0000 0000",
        "  RDFFIR                    (0x08011C00) C0DA 5009 0400 0000",
        "  RDFFIR_MASK               (0x08011C03) 0038 0FFF A000 7FFF",
        "  RDFFIR_ACT1               (0x08011C07) FFC3 7000 17E6 0000",
        "node 0 ocmb 62 (Explorer 2.0) ******************************",
        "  CHIPLET_OCMB_FIR_MASK     (0x08040002) 6627 FFE0 0000 0000"
    ]
},
"User Data 8": {
    "Section Version": "1",
    "Sub-section type": "1",
    "Created by": "0x2000",
    "PEL Internal Debug Data": {
        "SRC": [
            "No VPD found for /xyz/openbmc_project/inventory/system/chassis/motherboard/dimm2: sd_bus_call: xyz.openbmc_project.Common.Error.ResourceNotFound: The resource is not found."
        ]
    }
}
}

ecorderoibm, 2023-06-29:
It appears you are taking a lot of memory errors. I would re-evaluate the settings you are using for the DDR interface. I assume you have done signal integrity simulations and have determined those settings. Another option would be to probe them and see what quality the bus appears to have. I can't tell you what to put there, because it is very particular to your design and implementation

Grubby0624, 2023-06-30:
1. Today, I limited the memory frequency to 2666MHz (of course, I also used the SI parameter of 2666). When entering the OS, I was able to run the Stream test, andthere were no MCE errors, but the results were also unqualified:
    For example, there are four 2666MHz cables under Proc0, with a bandwidth of only about 20000MB/s
I currently want to address the issue of stream testing reporting MCE at 3200MHz. Based on the absence of MCE errors at 2666MHz, I think this may be an SI issue. I have many 3200MHz parameter combinations that can be used for training, and I plan to try running streams one by one to see if there are MCE errors. May I ask if there is a better method, or do you suspect that this may be related to which parameter settings? This way, I can focus on and test this parameter.
2. I received an SI simulation report from my colleague, and I understand that the recommended value for this ODT is related to DDR:
Operation type	Memory Controller configuration		DRAM configuration
Write			pm27_dq_d2828						DQ_ODT_80
Read			pm27_dq_t48							DQ_RON_34_ODT_OFF
Address			pm27_ca_d40_srf_c3p					DCA_IBT_100
But I'm not sure if what we mean by 'signal integrity simulations' is the same thing. sorry, Because I haven't read such simulation reports before, I'm not sure which results to focus on~

Grubby0624, 2023-07-17:
I change the IO_ OMI_ CHANNEL default value to LOSS to LOW_ LOSS, and then using different SI parameters on two different sets of DIMMs, the memory can now boot and run Streams without reporting MCE errors. Unfortunately, the Stream performance still does not meet the requirements

Grubby0624, 2023-07-17:
1. I only configure one DIMM under each Proc, and then specify 1/2/12 cores to run the Stream. The resulting values are set to 6000MB/s (the theoretical value should be 25GB/s)

2. When specifying a Core to run a Stream, I use the top command to check the CPU usage. The Stream application occupies 100% of the CPU

3. I put the ARRAY of the stream_ When the SIZE is downgraded and paired with 128K, the measured bandwidth of a core is approximately 490GB/s. I think this should be a memory operation directly completed in the cache.

May I ask:
    1. At this point, there are no PEL or other error reports. Can I assume that the memory subsystem has not reported any errors? If so, how should we monitor it?

    2. Regarding my above experimental phenomenon, can I lock the issue onto the OMI -->OCMB -->DDR link? What I mean is to exclude the impact of certain errors within the Processor?

dcrowell77, 2023-07-17:
1 - Yes, if there are no PELs then there have been no errors detected.
2 - I don't see any indication of "errors" so this is all just hardware tuning.

Have you been in contact with your IBM contacts regarding how the theoretical versus actual numbers relate from a hardware perspective?  The Hostboot team is not directly involved in any of the hardware tuning or performance measurements.

@ecorderoibm @steffenchris - Anything to add?

Grubby0624, 2023-07-31:
Regarding the issue of Stream performance not meeting standards, we found that there was an issue with the I2C address configuration of our platform's Thermal Sensor. We sent 'EXPFW_TEMP_SENSOR_CONFIG_INTERVAL_READ' to OCMB, but OCMB was unable to read a valid sensor value through the corresponding I2C address:
If we don't send EXP_ FW_ TEMP_ SENSOR_ CONFIG_ INTERVAL_ READ to OCMB, and the results of the Stream test meet the standards.
    https://github.com/open-power/ocmb-explorer-fw/blob/33e3301f619163c8d3aad373b045cb8db494d3d1/_exp/src/temp_sensor/temp_sensor_plat.c#L564C19-L564C19

By the way:
I guess OCMB will enter a "safe mode" if it cannot read the Thermal Sensor, but there is no relevant introduction found in the code and Spec. Can you provide a professional explanation? Or do you know where I can get relevant explanations?

Grubby0624, 2023-07-31:
Unfortunately, we encountered same MCE error while running HTX:
    1. If we specify the frequency to 3200MHz, running mdt.mem for less than 10 minutes will result in MCE error. If we specify the frequency to 2666MHz, there will be no MCE error even after running for 40 minutes. So I guess it's still a problem of Signal integrity
    2. Specify 3200MHz for Stream testing on the same platform and no MCE issues will be reported again
After an MCE error occurred, we triggered checkstop to collect FIRdata by injecting error to TI, and found that there were 2 MCs in each Proc that always reported the following error:
        {
            "Chip Desc": "node 0 proc 0 (P10 2.0)",
            "Signature": "MC_DSTL_FIR(1)[1] Subchannel A AFU initiated Recoverable Attention",
            "Attn Type": "recoverable"
        },
And when trying on different platforms, the MC numbers that reported errors are consistent. Regarding this error, I would like to ask:
    1. Based on the error content, can you determine whether this is an SI issue with the DDR bus or the OMI bus?
    2. What is the "AFU" mentioned here?
Regarding this MCE error, I plan to adjust the SI parameters of DDR and OMI. I would like to confirm what the following two OMI attributes mean:
    IO_ OMI_ PRE1 IO_ OMI_ PRE2
Our current values for both values are 0.
Do you have any other debugging suggestions?

dcrowell77, 2023-07-31:
> IO_ OMI_ PRE1 IO_ OMI_ PRE2

I can't personally tell you exactly what these are but I can say that we have them both set to zero on all of our boxes.

dcrowell77, 2023-07-31:
AFU = Accelerator Function Unit.  It was left over from OpenCAPI.  It is still referenced throughout OMI since its all OpenCAPI based.

Grubby0624, 2023-08-01:
Thank you for your answer. I now have some concepts about the problems I am facing.
So this is definitely used on the OMI interface, we cannot disable this module? My concern is that there are issues with both our OMI and DDR bus SI. I think I know how to adjust the SI parameters of DDR, but what are the SI parameters of OMI? Can we also adjust it?

steffenchris, 2023-08-01:
These values control the OMI Tx FFE.  Where each value is X/128 precursor.  For example, if IO_OMI_PRE1=8.  Then 8/128=6.25% of Tx FFE Precursor1.
IO_ OMI_ PRE1
IO_ OMI_ PRE2

dcrowell77, 2023-08-01:
> we cannot disable this module?

Note sure what you mean by "module" here.  The OMI bus is obviously fundamental to the system design.  You could try masking off the FIRs that are turning on but I suspect that will just hide the issue and you'll fail in some other way.

Grubby0624, 2023-08-07:
I put PHY_ The INIT mode has been set to WRITE_ EYE_ Train, I was puzzled by the printed results. Taking Rank0's bit0 as an example, from these logs, I understand that Vref values around 50 should be selected to ensure that as many Delays as possible can pass:
163.65588|ocmb:k0:n0:s0:p26 VrefDQ RANK 0, DBYTE 0 BIT 0 EYE INDEX 0 -- MIN: 127 MAX: 0
163.65589|ocmb:k0:n0:s0:p26 VrefDQ RANK 0, DBYTE 0 BIT 0 EYE INDEX 1 -- MIN: 127 MAX: 0
163.65590|ocmb:k0:n0:s0:p26 VrefDQ RANK 0, DBYTE 0 BIT 0 EYE INDEX 2 -- MIN: 44 MAX: 50
163.65591|ocmb:k0:n0:s0:p26 VrefDQ RANK 0, DBYTE 0 BIT 0 EYE INDEX 3 -- MIN: 40 MAX: 55
163.65592|ocmb:k0:n0:s0:p26 VrefDQ RANK 0, DBYTE 0 BIT 0 EYE INDEX 4 -- MIN: 34 MAX: 59
163.65593|ocmb:k0:n0:s0:p26 VrefDQ RANK 0, DBYTE 0 BIT 0 EYE INDEX 5 -- MIN: 33 MAX: 62
163.65595|ocmb:k0:n0:s0:p26 VrefDQ RANK 0, DBYTE 0 BIT 0 EYE INDEX 6 -- MIN: 29 MAX: 66
163.65596|ocmb:k0:n0:s0:p26 VrefDQ RANK 0, DBYTE 0 BIT 0 EYE INDEX 7 -- MIN: 23 MAX: 73
163.65597|ocmb:k0:n0:s0:p26 VrefDQ RANK 0, DBYTE 0 BIT 0 EYE INDEX 8 -- MIN: 21 MAX: 73
163.65598|ocmb:k0:n0:s0:p26 VrefDQ RANK 0, DBYTE 0 BIT 0 EYE INDEX 9 -- MIN: 17 MAX: 73
163.65599|ocmb:k0:n0:s0:p26 VrefDQ RANK 0, DBYTE 0 BIT 0 EYE INDEX 10 -- MIN: 18 MAX: 73
163.65600|ocmb:k0:n0:s0:p26 VrefDQ RANK 0, DBYTE 0 BIT 0 EYE INDEX 11 -- MIN: 18 MAX: 73
163.65602|ocmb:k0:n0:s0:p26 VrefDQ RANK 0, DBYTE 0 BIT 0 EYE INDEX 12 -- MIN: 17 MAX: 73
163.65603|ocmb:k0:n0:s0:p26 VrefDQ RANK 0, DBYTE 0 BIT 0 EYE INDEX 13 -- MIN: 18 MAX: 73
163.65604|ocmb:k0:n0:s0:p26 VrefDQ RANK 0, DBYTE 0 BIT 0 EYE INDEX 14 -- MIN: 17 MAX: 73
163.65605|ocmb:k0:n0:s0:p26 VrefDQ RANK 0, DBYTE 0 BIT 0 EYE INDEX 15 -- MIN: 17 MAX: 73
163.65606|ocmb:k0:n0:s0:p26 VrefDQ RANK 0, DBYTE 0 BIT 0 EYE INDEX 16 -- MIN: 17 MAX: 73
163.65607|ocmb:k0:n0:s0:p26 VrefDQ RANK 0, DBYTE 0 BIT 0 EYE INDEX 17 -- MIN: 18 MAX: 73
163.65609|ocmb:k0:n0:s0:p26 VrefDQ RANK 0, DBYTE 0 BIT 0 EYE INDEX 18 -- MIN: 20 MAX: 73
163.65610|ocmb:k0:n0:s0:p26 VrefDQ RANK 0, DBYTE 0 BIT 0 EYE INDEX 19 -- MIN: 22 MAX: 71
163.65611|ocmb:k0:n0:s0:p26 VrefDQ RANK 0, DBYTE 0 BIT 0 EYE INDEX 20 -- MIN: 23 MAX: 70
163.65612|ocmb:k0:n0:s0:p26 VrefDQ RANK 0, DBYTE 0 BIT 0 EYE INDEX 21 -- MIN: 24 MAX: 68
163.65613|ocmb:k0:n0:s0:p26 VrefDQ RANK 0, DBYTE 0 BIT 0 EYE INDEX 22 -- MIN: 27 MAX: 66
163.65614|ocmb:k0:n0:s0:p26 VrefDQ RANK 0, DBYTE 0 BIT 0 EYE INDEX 23 -- MIN: 32 MAX: 63
163.65616|ocmb:k0:n0:s0:p26 VrefDQ RANK 0, DBYTE 0 BIT 0 EYE INDEX 24 -- MIN: 33 MAX: 60
163.65617|ocmb:k0:n0:s0:p26 VrefDQ RANK 0, DBYTE 0 BIT 0 EYE INDEX 25 -- MIN: 38 MAX: 56
163.65618|ocmb:k0:n0:s0:p26 VrefDQ RANK 0, DBYTE 0 BIT 0 EYE INDEX 26 -- MIN: 39 MAX: 55
163.65619|ocmb:k0:n0:s0:p26 VrefDQ RANK 0, DBYTE 0 BIT 0 EYE INDEX 27 -- MIN: 45 MAX: 52
163.65620|ocmb:k0:n0:s0:p26 VrefDQ RANK 0, DBYTE 0 BIT 0 EYE INDEX 28 -- MIN: 127 MAX: 0
163.65621|ocmb:k0:n0:s0:p26 VrefDQ RANK 0, DBYTE 0 BIT 0 EYE INDEX 29 -- MIN: 127 MAX: 0
163.65623|ocmb:k0:n0:s0:p26 VrefDQ RANK 0, DBYTE 0 BIT 0 EYE INDEX 30 -- MIN: 127 MAX: 0
But it ultimately selects VrefDQ_ Center is 22:
    163.78035|ocmb:k0:n0:s0:p26 VrefDQ_Center RANK 0, NIBBLE 0: 22

Grubby0624, 2023-08-07:
May I ask:
1. Is there a deviation in my understanding of the training log? Why is the training result different from what I expected?
2. InitVrefDQ [0] is a parameter that affects WRITE_ EYE_ Does the process of training have any impact? I have modified this value, does the training result seem to have not changed?
3. If my previous understanding of the training results is correct, how should I change the Vref training results to around 50?
4. If PHY_ The mode of INIT is DEFAULT_ Train, how is its Vref determined?

dcrowell77, 2023-08-07:
@ecorderoibm @esteban012 can maybe weigh in on these questions.  You may need to engage Microchip as well since their firmware directly drives the DDR training, not Hostboot.

ecorderoibm, 2023-08-07:
The initVrefDQ setting will only tell the algorithm where to start. it really should have no impact on the final results. I think your opinion on the results is correct. The question as to why it selects a lower number than you expect, I would ask to MCHP. I am not sure where to look to know why, they will probably know. They will also be able to tell you how to override the training results if you are looking to try it at a different set point. Same for question #4 I think you can ask them to take you through the training algorithm, which will also be part of understanding why the Vref result was selected.

esteban012, 2023-08-07:
1. I am not sure of the details used. Microchip would need to respond to specific questions regarding the draminit training algorithm. However, you did not post the TxDqDly_Center. Could you please post the results for TxDqDly_Center for the bit mentioned?
2. InitVrefDQ is the initial VREF that is used for the write eye training. The algorithm should then train the WR VREF and write delays after using this setting as a starting point. If you want to skip WR VREF training, 
3. To update the WR VREF results, you would need to use CCS to issue the WR VREF latch commands on a per DRAM basis. Please consult the generic/memory/lib/dimm/ddr4/pda.H for the API on how to handle the latching commands on a per DRAM basis
4. I am not sure of the details used. Microchip would need to respond to specific questions regarding the draminit training algorithm.

Grubby0624, 2023-08-11:
1. The TxDqDly_Center is 15
2. When conducting MCBIST testing, we found that the memory that reported MCE errors would have the following errors:
 2023-08-11T15:02:29.532101698,[INF]:#### MCBIST: explorer:k0:n0:s0:p04 ####
2023-08-11T15:02:29.532117806,[INF]:########## PORT: 0 #############
2023-08-11T15:02:29.532120243,[INF]:Byte:        0011223344556677 8899
2023-08-11T15:02:29.532121884,[INF]:Nibble:      0101010101010101 0101
2023-08-11T15:02:29.532123572,[INF]:### DIMM: 0 ###
2023-08-11T15:02:29.532125242,[INF]:Mask        :0000000000000000 00ff
2023-08-11T15:02:29.532126816,[INF]:R0 Err Count:f............... ..MM
2023-08-11T15:02:29.532128386,[INF]:R0 Err Flags:X............... ..MM
2023-08-11T15:02:29.532129943,[INF]:R1 Err Count:ff.............. ..MM
2023-08-11T15:02:29.532131483,[INF]:R1 Err Flags:XX.............. ..MM
I will paste the corresponding error trap on the next reply. Regarding this error report, I would like to consult:
    1. Is this error related to MCE errors? Because we found that the memory location where the MCE error was reported corresponds to the memory location where this error was reported in MCBIST.
    2. What does MCBIST specifically do? I understand that it is to write a fixed pattern into DRAM through a method called super fast, and then read it out to determine whether the writing was successful? But what addresses have been written, and how is it determined whether they are successful?
    3. Are there any other suggestions from this log? I'm not sure if the MCE error is a signal integrity issue now?

dcrowell77, 2023-08-11:
@avalluriMu - fyi

Grubby0624, 2023-08-29:
Hello, thank you for your support. After modifying the values of the following registers, the issue of MCE error has been resolved:
https://github.com/open-power/hostboot/blob/fa90493edad80a0b34737cd83267712a88899ce1/src/import/chips/ocmb/explorer/common/include/explorer_scom_addresses.H#L1178
But I still have two questions and I hope you can help me answer them:
1. The interpretation of this register is "Read to Read to different master". What does the 'master; here mean?
2. I see that the register addresses related to OCMB are defined in these two files: explorer_ scom_ addresses.H explorer_ scom_ addresses_fixes.H, what does' fixes' mean here?Are there any differences in the SCOM addresses between these two files?
I think after understanding these two issues, we can close this issue.
Thank you very much!

ecorderoibm, 2023-08-29:
1. Master in this case is master rank, ie same chip select 
2. those registers have no real difference. the first file is the main one and the _fixes one are ones that were missing from the first one.

Grubby0624, 2023-08-31:
Got it. Thanks again!

closed 2023-08-31 by Grubby0624