open-power/hostboot #229: Small core CPU report checkstop error when wof enabled

lili-lilili, 2023-06-25:
We are testing small core CPU with OPAL.
After enableling wof, small core CPU will report checkstop error，but big core CPU will not.
There are roughly two types of checkstop errors, all of the report some IOHS_DLP_FIR_SMP error.

lili-lilili, 2023-06-25:
**1. When hostboot istep is execute，“TOD_ERROR(0)[21] S_PATH_1_STEP_CHECK_ERROR” checkstop reported.**
{
"Private Header": {
    "Section Version":          "1",
    "Sub-section type":         "0",
    "Created by":               "0xE500",
    "Created at":               "12/21/2022 20:07:14",
    "Committed at":             "12/21/2022 20:07:14",
    "Creator Subsystem":        "BMC",
    "CSSVER":                   "",
    "Platform Log Id":          "0x5000053B",
    "Entry Id":                 "0x5000053B",
    "BMC Event Log Id":         "5559"
},
"User Header": {
    "Section Version":          "1",
    "Sub-section type":         "0",
    "Log Committed by":         "0x2000",
    "Subsystem":                "Miscellaneous",
    "Event Scope":              "Entire Platform",
    "Event Severity":           "Unrecoverable Error",
    "Event Type":               "Not Applicable",
    "Action Flags": [
                                "Service Action Required",
                                "Report Externally",
                                "HMC Call Home"
    ],
    "Host Transmission":        "Not Sent",
    "HMC Transmission":         "Not Sent"
},
"Primary SRC": {
    "Section Version":          "1",
    "Sub-section type":         "1",
    "Created by":               "0xE500",
    "SRC Version":              "0x02",
    "SRC Format":               "0x55",
    "Virtual Progress SRC":     "False",
    "I5/OS Service Event Bit":  "False",
    "Hypervisor Dump Initiated":"False",
    "Backplane CCIN":           "2E2D",
    "Terminate FW Error":       "False",
    "Deconfigured":             "False",
    "Guarded":                  "False",
    "Error Details": {
        "Message":              "Error Signature: 0x20DA0020 0x00030001 0x7D9B0015"
    },
    "Valid Word Count":         "0x09",
    "Reference Code":           "BD70E510",
    "Hex Word 2":               "00080055",
    "Hex Word 3":               "2E2D0010",
    "Hex Word 4":               "CC009542",
    "Hex Word 5":               "00000000",
    "Hex Word 6":               "20DA0020",
    "Hex Word 7":               "00030001",
    "Hex Word 8":               "7D9B0015",
    "Hex Word 9":               "00000000",
    "Callout Section": {
        "Callout Count":        "1",
        "Callouts": [{
            "FRU Type":         "Maintenance Procedure Required",
            "Priority":         "Mandatory, replace all with this type as a unit",
            "Procedure":        "BMC0002"
        }]
    },
    "SRC Details": {
        "Primary Attention": "system checkstop",
        "Signature Description": {
            "Chip Desc": "node 0 proc 3 (P10 2.0)",
            "Signature": "TOD_ERROR(0)[21] S_PATH_1_STEP_CHECK_ERROR",
            "Attn Type": "checkstop"
[... 564 lines cut ...]
        "  CFIR_IOHS_RE              (0x1F040001) 8400 0000 0000 0000",
        "  IOHS_DLP_PHY_CONFIG       (0x1F01100C) AF00 0000 0000 00C1",
        "  IOHS_DLP_CONFIG           (0x1F01100A) 880F 0004 E200 F64B",
        "  IOHS_DLP_CONTROL          (0x1F01100B) C000 0000 C000 0000",
        "  IOHS_DLP_SEC_CONFIG       (0x1F01100D) 0000 0000 0000 001A",
        "  IOHS_DLP_OPTICAL_CONFIG   (0x1F01100F) 3878 053F 00E4 8C10",
        "  IOHS_DLP_LINK0_RX_LANE_CO (0x1F011012) 0000 0000 0FF8 0000",
        "  IOHS_DLP_LINK1_RX_LANE_CO (0x1F011013) 0000 0000 0FF8 0000",
        "  IOHS_DLP_LINK0_INFO       (0x1F011014) 316E 0000 0000 00EB",
        "  IOHS_DLP_LINK1_INFO       (0x1F011015) 316E 0000 0000 000A",
        "  IOHS_DLP_REPLAY_THRESHOLD (0x1F011018) 6FE0 0000 0000 0000",
        "  IOHS_DLP_SL_ECC_THRESHOLD (0x1F011019) 7FC0 0000 0000 0000",
        "  IOHS_DLP_LINK0_SYN_CAPTUR (0x1F011022) 252E E4B2 2020 0471",
        "  IOHS_DLP_LINK1_SYN_CAPTUR (0x1F011023) 2DA3 2DC6 A004 4001",
        "  IOHS_DLP_LINK0_QUALITY    (0x1F011026) 02F0 2F00 0000 0000",
        "  IOHS_DLP_LINK1_QUALITY    (0x1F011027) 0300 3000 0000 0000",
        "  IOHS_DLP_DLL_STATUS       (0x1F011028) 7F05 4A7F 0548 C000",
        "  IOHS_DLP_MISC_ERROR_STATU (0x1F011029) 0000 00CF 0000 0000",
        "  IOHS_DLP_FIR              (0x1F011000) F303 CF00 0000 0000",
        "  IOHS_DLP_FIR_MASK         (0x1F011003) FCFC 3FFF FCC0 C000",
        "  IOHS_DLP_FIR_ACT0         (0x1F011006) FCFC 3FFF FCC0 0000",
        "  IOHS_DLP_FIR_ACT1         (0x1F011007) FFFF FFFF FFFF FFFC"
    ]
}
}

lili-lilili, 2023-06-25:
**2. When the machine just boot to OS or we have just started stream test, "MC_DSTL_FIR(4)[22] Subchannel A Channel timeout" checkstop error will be report.**

{
"Private Header": {
    "Section Version":          "1",
    "Sub-section type":         "0",
    "Created by":               "0xE500",
    "Created at":               "12/21/2022 16:26:43",
    "Committed at":             "12/21/2022 16:26:43",
    "Creator Subsystem":        "BMC",
    "CSSVER":                   "",
    "Platform Log Id":          "0x500003E9",
    "Entry Id":                 "0x500003E9",
    "BMC Event Log Id":         "1414"
},
"User Header": {
    "Section Version":          "1",
    "Sub-section type":         "0",
    "Log Committed by":         "0x2000",
    "Subsystem":                "Memory Card/FRU",
    "Event Scope":              "Entire Platform",
    "Event Severity":           "Unrecoverable Error",
    "Event Type":               "Not Applicable",
    "Action Flags": [
                                "Service Action Required",
                                "Report Externally",
                                "HMC Call Home"
    ],
    "Host Transmission":        "Not Sent",
    "HMC Transmission":         "Not Sent"
},
"Primary SRC": {
    "Section Version":          "1",
    "Sub-section type":         "1",
    "Created by":               "0xE500",
    "SRC Version":              "0x02",
    "SRC Format":               "0x55",
    "Virtual Progress SRC":     "False",
    "I5/OS Service Event Bit":  "False",
    "Hypervisor Dump Initiated":"False",
    "Backplane CCIN":           "2E2D",
    "Terminate FW Error":       "False",
    "Deconfigured":             "False",
    "Guarded":                  "True",
    "Error Details": {
        "Message":              "Error Signature: 0x20DA0020 0x00030002 0xBCE50416"
    },
    "Valid Word Count":         "0x09",
    "Reference Code":           "BD24E510",
    "Hex Word 2":               "00080055",
    "Hex Word 3":               "2E2D0010",
    "Hex Word 4":               "CC009544",
    "Hex Word 5":               "01000000",
    "Hex Word 6":               "20DA0020",
    "Hex Word 7":               "00030002",
    "Hex Word 8":               "BCE50416",
    "Hex Word 9":               "00000000",
    "Callout Section": {
        "Callout Count":        "3",
        "Callouts": [{
            "FRU Type":         "Normal Hardware FRU",
            "Priority":         "Mandatory, replace all with this type as a unit",
            "Location Code":    "U78DA.ND0.WZS01SL-P0-C25",
            "Part Number":      "",
            "CCIN":             "",
            "Serial Number":    ""
        }, {
            "FRU Type":         "Normal Hardware FRU",
            "Priority":         "Mandatory, replace all with this type as a unit",
            "Location Code":    "U78DA.ND0.WZS01SL-P0-C24",
[... 765 lines cut ...]
        "  IOHS_DLP_SL_ECC_THRESHOLD (0x1F011019) 7FC0 0000 0000 0000",
        "  IOHS_DLP_LINK0_SYN_CAPTUR (0x1F011022) 8C39 67C2 F2A2 1901",
        "  IOHS_DLP_LINK1_SYN_CAPTUR (0x1F011023) 78EE 9F6C A009 0001",
        "  IOHS_DLP_LINK0_QUALITY    (0x1F011026) 02F0 3000 0000 8900",
        "  IOHS_DLP_LINK1_QUALITY    (0x1F011027) 0300 3000 0000 0000",
        "  IOHS_DLP_DLL_STATUS       (0x1F011028) 7F05 4A7F 0548 C000",
        "  IOHS_DLP_MISC_ERROR_STATU (0x1F011029) 0000 00CF 0000 0000",
        "  IOHS_DLP_FIR              (0x1F011000) F303 CF00 0000 0000",
        "  IOHS_DLP_FIR_MASK         (0x1F011003) FCFC 3FFF FCC0 C000",
        "  IOHS_DLP_FIR_ACT0         (0x1F011006) FCFC 3FFF FCC0 0000",
        "  IOHS_DLP_FIR_ACT1         (0x1F011007) FFFF FFFF FFFF FFFC"
    ]
},
"User Data 8": {
    "Section Version": "1",
    "Sub-section type": "1",
    "Created by": "0x2000",
    "PEL Internal Debug Data": {
        "SRC": [
            "No VPD found for /xyz/openbmc_project/inventory/system/chassis/motherboard/dimm24: sd_bus_call: xyz.openbmc_project.Common.Error.ResourceNotFound: The resource is not found.",
            "No VPD found for /xyz/openbmc_project/inventory/system/chassis/motherboard/dcm1/cpu1: sd_bus_call: xyz.openbmc_project.Common.Error.ResourceNotFound: The resource is not found."
        ]
    }
}
}

dcrowell77, 2023-06-26:
Please post the boot console as well, that provides more context on when in the boot the failure is happening.  I'm not familiar with #1 (TOD error) so I have to do some research.  Problem #2 (subchannel checkstop) seems unrelated to smallcore or WOF, you may just have a suspect DDIMM and/or DDIMM connector.  I would suggest you try (carefully) moving some parts around to see if it follows the slot or the DDIMM.

It is odd that WOF enablement affects anything during Hostboot because we don't really have the PM complex logic enabled while we run.

dcrowell77, 2023-06-26:
All of those SMP errors are likely the root cause of your issues (for at least 1, maybe 2).  I can tell from the output that the checkstop happens in istep 21.2  ("0x0000283c": "0xaa801502").  That is the point where we start the PM complex up.  Our guess is that this is related to the voltage issues that have already been discussed elsewhere.  We recommend taking a close look at all the voltages before, during, and after the checkstop to see if things look okay.

lili-lilili, 2023-07-08:
Based on more test cases, let me update the issue.

1. The checkstop with SMP errors may occur in istep 9.6/istep 11.3/istep 21.2 or OS.
So it seems that this issue is not directly related to WOF.
But if the machine can boot to OS (with wof enable) and we do some stream test, then the checkstop is likely to occur.

2. I am tried to reduce the link frequency  of IOHS to 25G(FREQ_IOHS_LINK_MHZ is modified from 32500M to 25781M)
The checkstop with SMP error will still happen.
But I'm not sure if this is the correct way to modify iohs link frequency.

3. In addition, test data shows that if a CPU experiences an SMP error for the first time, the probability of the CPU repeating the problem in subsequent tests will greatly increase. I am not sure if this is due to CPU hardware damage or if the CPU will record certain error internally(On P9, we have encountered situations where xbus errors are recorded inside the CPU).

4. Also, we encountered this issue when using a single CPU configuration.

5. Now, i am checking IO_IOHS_CHANNEL_LOSS attribute of IOHS.
Its setup is different on the different IOHS on the Rainier.
If the SPM interconnected is across the CPU, the attribute is set to HIGH_LOSS, otherwise the attribute is set to LOW_LOSS.
It seems that this attribute is related to SI,  I'm wondering if I should try changing it.

closed 2023-09-13 by lili-lilili