open-power/hostboot #231: About Simics

LoserJL, 2023-08-16:
Where to get Simics to run hostboot before there is actual hardware?

dcrowell77, 2023-08-17:
You would have to purchase a license from Wind River (https://www.windriver.com/products/simics) and also get Mambo from IBM (https://www.ibm.com/support/pages/power10-functional-simulator-related-resources).

All of the "standalone compile" toolchains should be available as part of gcc.  It is just a cross-compiler.  I would recommend using op-build to compile.  You can then install that toolchain from there to use independently if needed.

closed 2023-08-17 by dcrowell77

LoserJL, 2023-08-17:
> You would have to purchase a license from Wind River (https://www.windriver.com/products/simics) and also get Mambo from IBM [...]

Thank you