open-power/hostboot #235: About libconsole.so

LoserJL, 2023-10-11:
Hi, experts
Why is the console module compiled as BASE_ MODULES, but it is EXT_IMAGE in the g_exttaskinfolist?
[image]

Thanks!

LoserJL, 2023-10-13:
> Looks like a very old code bug.

I got it, thank you!

In VfsRp:_loadUnload function, if a specific module cannot be found in the extend module, it will be redetermined whether it is in the base module, so there will be no problem with execution

dcrowell77, 2023-10-13:
Yeah, the code functions correctly as written, but things look funny and are confusing.  We're working on a cleanup.