open-power/hostboot #239: Is there a register on P10 that can directly adjust "write delay values for one of the MEMINTDnnB pins"

Grubby0624, 2023-10-19:
I remember there is a register on P9 that directly adjusts the "write delay values for one of the MEMINTDnnB pins". Is there a similar register on P10 that can directly affect the shemoo test results

dcrowell77, 2023-10-19:
I'm not familiar with MEMINTDnnB.  Can you point me to some code that references it or tell me the specific scom address from P9?

Grubby0624, 2023-10-20:
Here are some combinations of registers whose addresses can be found in this file:
      src\import\chips\p9\common\include\p9_mc_scom_addresses.H
I am interested in the following two sets of registers:
     WR_DQ_DELAY_REG: https://github.com/open-power/hostboot/blob/74d0f6e3ff0bec6488715c37e12f1462239548c6/src/import/chips/p9/procedures/hwp/memory/lib/phy/dp16.C#L1078
      READ_DELAY_REG: https://github.com/open-power/hostboot/blob/74d0f6e3ff0bec6488715c37e12f1462239548c6/src/import/chips/p9/procedures/hwp/memory/lib/phy/dp16.C#L636

closed 2024-01-10 by Grubby0624