open-power/hostboot #240: About intvect_system_reset_external

LoserJL, 2023-10-26:
Hi, 
As far as I know, GPRS of problem in the nap state is lost, but in the intvect_system_reset_external, intvect_external is called, and the kernel_save_task is called in intvect_external, so the incorrect task status will be saved, this results in a serious error when returning the task

[image]

In fact, when I actually run a task, it enters nap:
            asm volatile("li 3, %0; sc 2" :: "K" (MISC_CPUNAP) : "r3", "cc");
            nap();
and then wakes up, when it return to the task, there is indeed a Data Storage exception:
            tid = 52 (my task)
            DAR = 0x28
            DSISR = 0x42000000
            nip = 0x7e564 (ld r2, 0x28(r1))
It can be seen that r1=0 did not recover normally

Could you please help me check if there was a mistake in my understanding? 

Thank you！

dcrowell77, 2023-10-30:
What is the context of when you are running this?  For most of the IPL the power management logic isn't even enabled so the threads/cores never lose any state.  After the PM complex is enabled we enable special wakeup to prevent any power management changes as well.  Are you doing something specific to avoid that?  I suspect none of this nap code has actually executed in a functional way since P8.

LoserJL, 2023-10-30:
> What is the context of when you are running this? For most of the IPL the power management logic isn't even enabled so the threads/cores never lose any state. After the PM complex is enabled we [...]

Yes, I was executing nap in a task in hostboot's user space for the purpose of testing the processor's nap functionality and found this problem, this is just my tests.
As you said, hostboot hasn't performed the entire function from entering the nap to waking up, and normally, hostboot doesn't have a need to enter the nap either. 
From the functionality of hostboot, this should not be considered an issue.
Thank you!

closed 2023-10-30 by LoserJL