open-power/hostboot #243: Where is the variable "iv_occsStarted" set to true during the runtime for P10 opal?

Theo0208, 2024-03-29:
The “processOccError” function in Hostboot for P10 needs to ensure that "iv_occsStarted" is true before continuing to handle OCC errors, but the iv_occsStarted is false during runtime. It seems that only phyp can set iv_occsStarted to true during runtime, so I want to konw where the variable iv_occsStarted is set to true during runtime for P10 opal.Thanks a lot.

Theo0208, 2024-04-03:
Excuse me, could I understand it this way:
Because PHYP system will not load and start PMC in the "call_host_runtime_setup" function during IPL,
PMC can only be loaded and started by calling "load_start_pm_complex" through "callSetupPMCLoadStartCallback" in the "rt_start" function during runtime.
But OPAL system will call "loadAndStartPMAll" and "processOccStartStatus" in the "call_host_runtime_setup" function during IPL, so for OPAL, it is unnecessary and will not to call "load_and_start_pm_complex" and "processOccStartStatus" function again during runtime, and this is the reason why the variable iv_occsStarted is false for P10 OPAL.

cjcain, 2024-04-03:
I am not sure if that last comment had a question or was just a statement.
As I said, that HTMGT::processOccStartStatus() function is what sets the iv_occsStarted flag.

You do not need to call the load_and_start_pm_complex() function.
You can add a direct call to HTMGT::processOccStartStatus() (similar to what was done in load_and_start_pm_complex)

Theo0208, 2024-04-15:
Okay,thanks.

closed 2024-04-24 by Theo0208