open-power/hostboot #244: Will the check_reset_count function of P10 be periodically called?

Theo0208, 2024-04-23:
According to the description in the Hostboot code, function "check_reset_count" is called periodically (OCC_RCOUNT_RESET_TIME), OCC_RCOUNT_RESET_TIME represents a time period of one hour.
      But the actual situation is that after the P10 machine is turned on, check_reset_count will not  be periodically called.
      How is the check_reset_count function called periodically in the code?
      Thanks a lot.

cjcain, 2024-04-23:
HTMGT is using the clock time (vs counter) so it does not need to be called regularly.
When the HTMGT code gets woken up (via OCC interrupt to report error or request reset or debug command), the timer will be checked to determine if the counts need to be reset. That check happens first, so if has been more than one hour since the prior wakeup, HTMGT would see that the timer has indeed expired and it will clear the reset counts.

Theo0208, 2024-04-26:
Okay, thanks.

closed 2024-04-26 by Theo0208