open-power/hostboot #245: Does the FSP based system support the attribute override

Grubby0624, 2024-04-24:
https://github.com/open-power/hostboot/blob/62817231aba75a3cdeb2bacf173fb8e90fde0357/src/usr/fapi2/plat_attr_override_sync.C#L528
I want to modify some ATTRs through attribute override so that I can perform some interesting DIY without flashing the firmware. Do FSP based systems also have similar mechanisms? If so, how can I modify the attribute of the FSP based system?
Thank you!

dcrowell77, 2024-04-24:
Yes, the HWSV component on the FSP consumes a text file of attribute overrides.  The Hostboot support is modeled after that logic that has been in place for many years.  The file goes into /tmp/FspOverrideAttrs.txt on the FSP.

Grubby0624, 2024-04-25:
Okay! So I can DIY in the FSP base system, thank you!!

closed 2024-04-25 by Grubby0624