open-power/hostboot #246: Will Mcbist check all addresses of all DIMMs?

Grubby0624, 2024-05-14:
https://github.com/open-power/hostboot/blob/c4227d1e3dcf86b00ce0275cfebe2148d2035a24/src/import/generic/memory/lib/utils/mcbist/gen_mss_memdiags.H#L678
Will Mcbist check overwrite all memory cells? Also, is the error detected by Mcbist during the IPL process handled by the prdf module?
Thanks!

sglancy6, 2024-05-14:
memdiags will initialize all addresses with good ECC and will verify all addresses have good ECC.

sglancy6, 2024-05-14:
Yes - PRD will log ECC errors found during memdiags. However logging of CEs in memdiags requires enabling mnfg policy flag 'memdiags CE screening'.

Grubby0624, 2024-05-23:
So, if memodiag detects the same type of error(Maintenance IMPE for example) on different memory cells, will it record all incorrect addresses? Will each address also be processed once during PRDF processing?
What I want to do now is to record all memory errors on the system, whether they are recoverable or unrecoverable.

Grubby0624, 2024-05-27:
Okay，got it, thank you very much!

closed 2024-05-27 by Grubby0624