open-power/hostboot #251: What is the purpose of this commit

Grubby0624, 2024-07-29:
https://github.com/open-power/hostboot/blob/1e3984c6540ba6d327d0c6f7aaccd783647ef27f/src/import/chips/ocmb/common/procedures/hwp/pmic/lib/utils/pmic_enable_utils.C#L476

I have a few questions about this commit: 43fd55aa243fe570b989da8f8ff9fcb893c7d218 "Added recovery procedure to DDR4 pmic_n_made_detect":
    1. What is the purpose of this commit?  If I don't add this commit and restart the machine after setting the PMIC output voltage bias, it can start up normally. However, if I add this commit, the I2C link to this PMIC will report an I2C No ACK error after booting up
    2.  I am using TPS53830 PMIC, and I understand that RA3 represents the I2C address for this model of PMIC. Why is its value changed to 0x00 here?
Thanks!!

dcrowell77, 2024-08-06:
I think there was a typo in your commit hash above.  Did you mean 43fd55aa243fe570b989da8f8bf9fcb893c7d218?  If so, the commit is adding some logic to recover a dead PMIC at runtime.  It has been working fine for us.  Where in the IPL do you see the failure?

I'm asking around about the second question.  If that is a different PMIC model than we use then our code may not support it.  In our case RA3 seems to be just a scratch register with no functional use.

dcrowell77, 2024-08-06:
Are you testing with DDIMMs or some other configuration?  Do you have DDR4 or DDR5 memory? It appears that you are going down a DDR4 code path using a PMIC that we only support on the DDR5 DDIMMs.  There are some implicit assumptions about which PMIC models are supported for which DDR versions that you seem to be violating.